MiczFlor/RPi-Jukebox-RFID #2222: No Skip Rewind Pause Stop Option available?

kichilron, 2024-01-28:
### Describe your problem 

I just upgraded one my old installations to a future3 installation and was curious to find no ability to bind RFID cards to skip, rewind, pause or stop. Is this intentional, is it going to be coming or am I missing something?

#### What's your hardware set up? 

RPi5, latest future3 branch, standard installation with installation script; everything working as far as I can tell.

s-martin, 2024-01-28:
Hi,

in future3 you can bind RFID cards to every RPC command, but you have to edit `cards.yaml` by yourself.

Please see here https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/future3/main/documentation/builders/card-database.md

If you have any further questions, just ask.

s-martin, 2024-01-28:
If you register a new card in the Web App there’s also an option to register it with commands in the drop-down menu.

directly editing the settings file shouldn’t be necessary.

kichilron, 2024-01-28:
Cannot confirm that I have these options. In the older version they were available.

I am comfortable with editing the cards.yaml - where can I find all available RPC calls?

s-martin, 2024-01-28:
If you run the `run_rpc_tool.py help` you should get all rpc commands.
See also here https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/future3/main/documentation/developers/coreapps.md#rpc

Be aware that you need to activate a venv before running a Python command, see https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/future3/main/documentation/developers/pyhton.md

AlvinSchiller, 2024-01-30:
These options have also been added recently on the future3/develop branch and will be included in the next release 3.5.0, comin soon 😁 
-> https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/2179

kichilron, 2024-01-30:
This is very good news @AlvinSchiller  - thank you!

AlvinSchiller, 2024-02-04:
Version 3.5.0 is released

closed 2024-02-04 by AlvinSchiller