MiczFlor/RPi-Jukebox-RFID #2224: Library view on web app can navigate above audiofolders directory on future3

thecyborganizer, 2024-01-31:
### Describe your problem 

In the web app, if you choose the "folders" view, once you start navigating into the directory structure, you can use the "back" button to go above the `audiofolders` directory, all the way up to the home directory of the user.

This exposes the whole filesystem of the host device to the web app, which might not always be desirable.

See below for a screen recording demonstrating this behavior:

https://github.com/MiczFlor/RPi-Jukebox-RFID/assets/3578699/ded9b5a4-508e-4281-8a8d-c4f6269ceff0

#### What's your hardware set up? 

Raspberry Pi Zero 2 W running 3.4.0

#### If possible, try to attach logs from ... (paths from RPi)

  * `~/RPi-Jukebox-RFID/shared/logs` -> General Jukebox logs
  * `~/INSTALL-XXXXXXXXX.log` -> The logfile being generated when installing the Jukebox code

s-martin, 2024-03-04:
Sorry for catching up so late.
Can you add the screencast again? At least I can’t play it.

AlvinSchiller, 2024-03-13:
Additional information: 
If you navigate in an empty folder, there is no "back" button anymore to navigate back.
You need to switch pages to come back to the top level view

s-martin, 2024-03-19:
@thecyborganizer 
This should be fixed now on future3/develop.

Could you check this?

pabera, 2024-04-08:
Problem was fixed

PR #2293 (2024-03-16): fix: Stop Back Action at audiofolder root level
This also allows to go back when folder is empty

Fixes #2224
commit 4d05c756 (2024-03-16): fix: Stop Back Action at audiofolder root level (#2293)

* bugfix: Stop Back Action at audiofolder root level

This also allows to go back when folder is empty

Fixes #2224

* fix: Flake8 issue

* Remove some minor issues
closed 2024-04-08 by pabera