MiczFlor/RPi-Jukebox-RFID #2248: 🐛 | RC522 PlaceNotSwipe broken with Full Id Support (develop)

AlvinSchiller, 2024-02-07:
## Bug

### What I did

Acrtivate full id support for Rfid Reader Rc522 introduced with https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/2214.
Activated "PlaceNotSwipe" function for reader.

### What happened

Playback starts when card is presented, but does not stop if card is removed.

### I expected this to happen

The playback has to stop if the card is removed, like it works in "legacy id" mode.

### Further information that might help

Introduced with feature https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/2214

## Software

### Base image and version

2023-12-05-raspios-bullseye-armhf-lite.img.xz

### Branch / Release

v2.6.0-alpha (develop)

### Installscript

`scripts/installscripts/install-jukebox.sh` with setup of rfid reader an "non legacy mode" for id.

## Hardware

### RaspberryPi version

3 A+

### RFID Reader

RFC522

AlvinSchiller, 2024-02-07:
@miohna There seems to be a bug with the full id support. Do you have a chance to look into it again? Probably the read function does something different.

s-martin, 2024-02-07:
Does place not swipe work for the future3 implementation?

AlvinSchiller, 2024-02-07:
Hmm good question... Have to test it

AlvinSchiller, 2024-02-08:
> A suggestion for the time being is to place a `reader.reader.device.wait_for_tag()` in the `if swipe_or_place == "PLACENOTSWIPE"` statement of file `scripts/daemon_rfid_reader.py` and remove it from the `readCard_legacy` method.

I think thats not possible, as the `wait_for_tag` is specific for the RC522 device and not present in other readers.
The `daemon_rfid_reader.py`uses the generic Reader class only defining `readCard` for all readers.

AlvinSchiller, 2024-02-08:
> Does place not swipe work for the future3 implementation?

"Place not Swipe" is working in v3.

miohna, 2024-02-08:
> [Verfolgungsbild][https://github.com/notifications/beacon/AL55YR2ZEMD522CTI2AEZCLYSRUYPA5CNFSM6AAAAABC5QEVZSWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTTTHXAII.gif]

I see, you are right. `wait_for_tag()` needs to go in the `readCard` method.

Maybe designed like in v3.

closed 2024-02-10 by AlvinSchiller