MiczFlor/RPi-Jukebox-RFID #2260: cheap RFID-chip does not work with RC522 module

Lippsson, 2024-02-15:
### Describe your problem 

I have 3 different types of RFID-Chips:  The card, and tag - which is included when buy a RC522 card reader and a cheap one you can see on that picture:
[image]

The card and tag works fine. But the cheaper chip did not recognized by the phoniebox software.
I tested the script from: https://pimylifeup.com/raspberry-pi-rfid-rc522/
which give me following feedback with the cheap tag:
[image]

#### What's your hardware set up? 
Pi3b 1GB
RC522 reader
USB reader
OLED Display 1.3"
MupiHat 

#### If possible, try to attach logs from ... (paths from RPi)
[errors.log.zip](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/14295322/errors.log.zip)
[INSTALL-1707596174.log](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/14295325/INSTALL-1707596174.log)

AlvinSchiller, 2024-02-15:
Hi, seems the Tag as some kind of authorization set, which we don't support.
Its also reported here https://github.com/pimylifeup/MFRC522-python/issues/31 , with a workaround hack supplied.

I saw you activated "legacy_mode" fpr RC522. Can you also try with the normal read mode?
This setting can be changed in the `rfid.yaml`

pabera, 2024-02-15:
Some chips don't work with RC522. They must support 13.56 Mhz. Other product numbers related to that are S50 or S70. [Here is an example](https://www.amazon.de/YARONGTECH-Classic%C2%AE-Selbstklebende-R%C3%BCckseite-Medaille/dp/B01MY0WK8R).

Another example: If you buy the [Neuftech reader](https://www.amazon.de/Neuftech-Reader-Kartenleseger%C3%A4t-Kartenleser-Kontaktlos/dp/B018OYOR3E), it is running at 125 Mhz and cards/chips working with it, won't work with RC522.

AlvinSchiller, 2024-02-15:
Do you now the exact name/type of the cheap tags?

Lippsson, 2024-02-15:
I ordered this one: https://www.aliexpress.com/item/32654646206.html?spm=a2g0o.order_list.order_list_main.91.335d1802BhMpY8

I tried both with legacy_mode Yes and No... did not had any effect...
Untorturable my programming knowledge is to bad to find the right place to paste the workaround.

AlvinSchiller, 2024-02-15:
> I ordered this one: https://www.aliexpress.com/item/32654646206.html?spm=a2g0o.order_list.order_list_main.91.335d1802BhMpY8

The link ends in a 404 - Page not found

Lippsson, 2024-02-15:
Sorry that was the link from my baset.

[image]
Here comes another link: https://www.aliexpress.com/item/32654646206.html?spm=a2g0o.productlist.main.1.3c1443f03ZWYEi&algo_pvid=fd2414be-be98-42ec-96b2-40723634eb35&algo_exp_id=fd2414be-be98-42ec-96b2-40723634eb35-0&pdp_npi=4%40dis%21SEK%21172.07%21137.64%21%21%2115.94%2112.75%21%40211b600817079948957251186ee66f%2112000028027041802%21sea%21SE%213335247857%21&curPageLogUid=rroNMg1J0AyM&utparam-url=scene%3Asearch%7Cquery_from%3A

owntheweb, 2024-04-13:
I picked up an RFID-RC522 recently and just learned about this project (neat!), also purchasing sticker. I'm designing a box currently (will share) and will take a look in about one to two weeks. I'll post an update on how my stickers worked (or didn't), testing separately from RPi-Jukebox-RFID first.

In general though, the RC522 reader seems to be limited on what it can read. No NFC tags supported to my understanding. Those are newer. I'm going to try these stickers (IMPORTANT: UNTESTED):
https://www.amazon.com/YARONGTECH-50pcs-MIFARE-Classic-adhesive-Sticker/dp/B01LZYOR7P?th=1

s-martin, 2024-04-15:
Moving this to Discussions as it is more a question.