MiczFlor/RPi-Jukebox-RFID #2271: Failed to build pyzmq

edednedy, 2024-02-18:
Raspberry pi 3 . Both stable and pre-release rpi-jukebox-rfid V3 throw off this error when installing.
A non-recoverable error occurred
Check Install Log for details
 Log:
ERROR: pyzmq is not installed
CMake build failed
error< subprocess-exited-with-error
Building wheel for pyzmq (pyproject.toml) did not run successfully
exit code 1

Faled to build pyzmq

s-martin, 2024-02-19:
Can you please post the full installation log?

edednedy, 2024-02-19:
I can't. After several hours of messing with it. Getting the same error. I reinstalled Raspberry Pi OS and tried RPI-Jukebox-RFID Version 2 and it installed all the way and seems to be working. I'm not sure what pyamg does or why it's causes Ver3 to crash on installation, Maybe it's not installed in Ver2.
If I can find another card, I may install Ver3 on it just so I can get back to the error installation log.

AlvinSchiller, 2024-02-19:
Version 2 does not use pyzmq.

AlvinSchiller, 2024-02-19:
It seems the new prerelease of pyzmq breaks installation.
I created a Bugreport https://github.com/zeromq/pyzmq/issues/1937

Hotfix for the installation will come soon (using last release version of pyzmq).

edednedy, 2024-02-19:
I originally got this error on installation of the stable Version 3. Then I tried the pre-release version. Same issue with both versions. If it is the newest pre-release that breaks it, shouldn't the stable version 3 work? Also, since version 2 works without it, what does it do?

AlvinSchiller, 2024-02-19:
Its the version of the pyzmq lib which causes the error. So version 3 stable and pre of this project are both affected.

Version 3 is a complete rewrite, so it uses a different architecture and libs.
Thats why version 2 is not affected, as it just does not use pyzmq.

AlvinSchiller, 2024-02-19:
Fixed with #2272

closed 2024-02-19 by AlvinSchiller

IgorWalton, 2024-04-15:
Tried a fresh install of 3.5.2 stable today, but the pyzmq installation still failed. Tried to use different SD cards without success.
Logs:
[INSTALL-1713209396.log](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/14984054/INSTALL-1713209396.log)

AlvinSchiller, 2024-04-15:
Hi, the logs state the Installation is run as user root.
How did you start the installation? 
Have you followed the install description https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/future3/main/documentation/builders/installation.md#stable-release?

And if this problem persists could you please open a new issue? New comments in closed issues get overlooked easily.

s-martin, 2024-04-16:
> EDIT:
> I just saw the ci also fails. pyzmq==26.0.0 was released a few hours ago, and it seems there are incompatibilities with bullseye...

So should pin the version of pyzmq

IgorWalton, 2024-04-16:
> Hi, the logs state the Installation is run as user root. How did you start the installation? Have you followed the install description https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/future3/main/documentation/builders/installation.md#stable-release?
> 
Yes, I followed the install description and named the user root.

pabera, 2024-04-16:
Let's follow up on this new issue here: #2344

PR #2272 (2024-02-19): Hotfix for pyzmq installation
fixes #2271
pyzmq installation fails due to the latest prerelease.

Fix: removed the `--pre` argument to use the latest release, which is known to work.