MiczFlor/RPi-Jukebox-RFID #2278: 🐛 | cant register multiple reader

Lippsson, 2024-02-28:
## Bug
i run the installation and tried to register more than one reader:
[image]

### What I did

### What happened
after i pressed y on the question "Do you want to add another device" the scipt endet.

### I expected this to happen

to be able to register another device.
I also tried to rename the RegisterDevice.py.Multi and run this script without success

### Base image and version

PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

Refresh index: 100% (390/390), done.
On branch master
Your branch is up to date with 'origin/master'.

Hardware: 

Pi3 B+
RC522
USB reader

AlvinSchiller, 2024-02-28:
Can you try answer with a capital "Y"?
From the code this is the needed answer...

Lippsson, 2024-02-28:
Ok. Thanks. That worked.. 
But unfortunable this took me to another issue. When i register only 1 Reader - in my case the RC522 - the Reader works fine. If i want to register 2 Reader (RC522 & USB Reader) the USB Reader works, but not the RC522

Lippsson, 2024-04-02:
Sorry! I have to come back while having issues to register RC522 and USB-Reader att the same time.

When i run the RegisterDevice.py.Multi script as RegisterDevice.py. I can register both readers but only the USB reader will work: 
[image]
The only way i can make the RC522 to work is to runt the setup_rc522.sh file in RPi-Jukebox-RFID/components/rfid-reader/RC522 folder, but than i could not use the USB reader

AlvinSchiller, 2024-04-26:
Checked it too and it seems like RC522 (and maybe other Non-USB Reader) are not functional with the multi reader setup :/

AlvinSchiller, 2024-04-26:
The problem with the possible answers to the continue questions will be fixed with https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/2357