MiczFlor/RPi-Jukebox-RFID #2279: 🐛 | Issue when i toggle between Album and Folder view

Lippsson, 2024-02-29:
I did a fresh future3/main installation on a bullseye os. Pi3
I added 1064 music files in 97 folder no folder or file has cover pictures embedded or as separate file

As you see in my recorded video all folder show up on the website. 

I toggle between Album and folder view. After 3-4 times it takes longer time to show all folders and about the 6th time, the website will stop working. The Video shows my actions after a reboot:

https://github.com/MiczFlor/RPi-Jukebox-RFID/assets/48052340/0c1af50d-3f14-4e30-bc18-3462a7fa972a

[errors.log](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/14449325/errors.log)
[app.log](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/14449326/app.log)

s-martin, 2024-03-02:
Maybe related to #2203 

Switching back and forth could reveal some problems with already existing data in the albumart cache not considered enough.

pabera, 2024-03-15:
The implementation of Cover Art is the bottleneck. It does not run in a separated thread. When you initially load your Library, the script goes through all albums. It's already taking quite a bit for a few albums. I can imagine that 1k albums might give your Pi a hard time. 

We have to revisit this once again and maybe allow the Cover Art Process to run independently from the main thread.

Lippsson, 2024-03-16:
I understand. What if we skip the covers completely and just load/show the cover on the player?

Lippsson, 2024-03-16:
Can you check with @AlvinSchiller ... He did also some tests to skip the covers but we had the issue anyway: https://www.youtube.com/watch?v=vHcGA-GywVc

s-martin, 2024-04-08:
#2325 has been merged.
You could check on future3/develop, if your issue is fixed.

s-martin, 2024-04-15:
#2325 fixed this on future3/develop.

Closing. Feel free to reopen, if your issue still persists.

closed 2024-04-15 by s-martin