MiczFlor/RPi-Jukebox-RFID #2285: 🐛 | RIFD cards not recognized fresh new installation

dvalob, 2024-03-03:
Hi

I installed a fresh new phoniebox, using the one line command and v 2.5.0 - 4101984b - master. I have rhe USB RFID reader configured from the first installation steps. When I swipe, I have the bip from the reader, but I do not see the card ID in the webinterface. Also, I do not have the latestID.txt file in the shared folder.

this sounds a bit weird. Any advice please? thanks!

 
my phoniebox-rfid-reader service is up.
-->

## Software

### Base image and version

 2.5.0 - 4101984b

Raspbian GNU/Linux 10 (buster)

## Hardware

### RaspberryPi version

3B 1.2

### RFID Reader

Bus 001 Device 004: ID 1a86:dd01 QinHeng Electronics 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1a86 QinHeng Electronics
  idProduct          0xdd01 
  bcdDevice            0.00
  iManufacturer           1 RFID Reader
  iProduct                2 RFID Reader
  iSerial                 3 ID READER
  bNumConfigurations      1

s-martin, 2024-03-03:
What's the output of `tail -n 200 /var/log/syslog`?

If you use Raspberry Pi OS "Bookworm", please use the command `journalctl -n 200`.

dvalob, 2024-03-03:
please see the output here
Mar  3 11:29:58 raspberrypi avahi-daemon[383]: Registering new address record for 192.168.1.124 on wlan0.IPv4.
Mar  3 11:29:58 raspberrypi dhcpcd[403]: wlan0: adding route to 192.168.1.0/24
Mar  3 11:29:58 raspberrypi dhcpcd[403]: wlan0: adding default route via 192.168.1.254
Mar  3 11:29:58 raspberrypi dhcpcd[403]: forked to background, child pid 568
Mar  3 11:29:58 raspberrypi systemd[1]: Started dhcpcd on all interfaces.
Mar  3 11:29:58 raspberrypi systemd[1]: Reached target Network.
Mar  3 11:29:58 raspberrypi systemd[1]: Starting Permit User Sessions...
Mar  3 11:29:58 raspberrypi systemd[1]: Reached target Network is Online.
Mar  3 11:29:58 raspberrypi systemd[1]: Starting Lighttpd Daemon...
Mar  3 11:29:58 raspberrypi systemd[1]: Starting /etc/rc.local Compatibility...
Mar  3 11:29:58 raspberrypi systemd[1]: Starting LSB: exim Mail Transport Agent...
Mar  3 11:29:59 raspberrypi systemd[1]: Started Phoniebox GPIO Control Service.
Mar  3 11:29:59 raspberrypi systemd[1]: Started Phoniebox Idle Watchdog Service.
Mar  3 11:29:59 raspberrypi systemd[1]: Starting OpenBSD Secure Shell server...
Mar  3 11:29:59 raspberrypi systemd[1]: Started Phoniebox RFID-Reader Service.
Mar  3 11:29:59 raspberrypi systemd[1]: Condition check resulted in fast remote file copy program daemon being skipped.
Mar  3 11:29:59 raspberrypi systemd[1]: Starting Music Player Daemon...
Mar  3 11:29:59 raspberrypi systemd[1]: Starting Samba NMB Daemon...
Mar  3 11:29:59 raspberrypi systemd[1]: Started Permit User Sessions.
Mar  3 11:29:59 raspberrypi systemd[1]: Started /etc/rc.local Compatibility.
Mar  3 11:29:59 raspberrypi systemd[1]: Started Getty on tty1.
Mar  3 11:29:59 raspberrypi systemd[1]: Reached target Login Prompts.
Mar  3 11:29:59 raspberrypi systemd[1]: Started OpenBSD Secure Shell server.
Mar  3 11:30:00 raspberrypi lighttpd[570]: 2024-03-03 11:29:59: (server.c.1493) WARNING: unknown config-key: deflate.cache-dir (ignored)
Mar  3 11:30:00 raspberrypi systemd[1]: Started Lighttpd Daemon.
Mar  3 11:30:00 raspberrypi exim4[572]: Starting MTA: exim4.
Mar  3 11:30:00 raspberrypi systemd[1]: Started LSB: exim Mail Transport Agent.
Mar  3 11:30:01 raspberrypi PhonieboxGPIOControl[574]: INFO:__main__:GPIO Started
Mar  3 11:30:01 raspberrypi PhonieboxGPIOControl[574]: INFO:__main__:['VolumeControl', 'PrevNextControl', 'PlayPause', 'Shutdown', 'PauseShutdown', 'Volume0', 'VolumeUp', 'VolumeDown', 'NextSong', 'PrevSong', 'FastForward', 'Rewind', 'Halt', 'RFIDDevice']
Mar  3 11:30:01 raspberrypi PhonieboxGPIOControl[574]: INFO:__main__:Device VolumeControl not enabled
Mar  3 11:30:01 raspberrypi PhonieboxGPIOControl[574]: INFO:__main__:Device PrevNextControl not enabled
Mar  3 11:30:01 raspberrypi PhonieboxGPIOControl[574]: INFO:__main__:Device PlayPause not enabled
Mar  3 11:30:01 raspberrypi PhonieboxGPIOControl[574]: INFO:__main__:Device Shutdown not enabled
Mar  3 11:30:01 raspberrypi PhonieboxGPIOControl[574]: INFO:__main__:Device PauseShutdown not enabled
Mar  3 11:30:01 raspberrypi PhonieboxGPIOControl[574]: INFO:__main__:Device Volume0 not enabled
Mar  3 11:30:01 raspberrypi PhonieboxGPIOControl[574]: INFO:__main__:Device VolumeUp not enabled
Mar  3 11:30:01 raspberrypi PhonieboxGPIOControl[574]: INFO:__main__:Device VolumeDown not enabled
Mar  3 11:30:01 raspberrypi PhonieboxGPIOControl[574]: INFO:__main__:Device NextSong not enabled
Mar  3 11:30:01 raspberrypi PhonieboxGPIOControl[574]: INFO:__main__:Device PrevSong not enabled
Mar  3 11:30:01 raspberrypi PhonieboxGPIOControl[574]: INFO:__main__:Device FastForward not enabled
Mar  3 11:30:01 raspberrypi PhonieboxGPIOControl[574]: INFO:__main__:Device Rewind not enabled
Mar  3 11:30:01 raspberrypi PhonieboxGPIOControl[574]: INFO:__main__:Device Halt not enabled
Mar  3 11:30:01 raspberrypi PhonieboxGPIOControl[574]: INFO:__main__:Device RFIDDevice not enabled
Mar  3 11:30:01 raspberrypi PhonieboxGPIOControl[574]: INFO:__main__:Ready for taking actions
Mar  3 11:30:01 raspberrypi systemd[1]: Started Samba NMB Daemon.
Mar  3 11:30:01 raspberrypi systemd[1]: Starting Samba SMB Daemon...
Mar  3 11:30:01 raspberrypi daemon_rfid_reader.py[579]: 2024-03-03 11:30:01,536 - root - INFO - Dir_PATH: /home/pi/RPi-Jukebox-RFID/scripts
Mar  3 11:30:02 raspberrypi systemd[1]: Started Samba SMB Daemon.
Mar  3 11:30:03 raspberrypi mpd: zeroconf: No global port, disabling zeroconf
Mar  3 11:30:03 raspberrypi systemd[1]: Started Music Player Daemon.
Mar  3 11:30:03 raspberrypi systemd[1]: Starting Phoniebox Startup...
Mar  3 11:30:03 raspberrypi bash[863]: Phoniebox is starting...
Mar  3 11:30:03 raspberrypi bash[863]: 2.5.0
Mar  3 11:30:03 raspberrypi bash[863]: AUDIOFOLDERSPATH="/home/pi/RPi-Jukebox-RFID/shared/audiofolders"
Mar  3 11:30:03 raspberrypi bash[863]: PLAYLISTSFOLDERPATH="/home/pi/RPi-Jukebox-RFID/playlists"
Mar  3 11:30:03 raspberrypi bash[863]: SWIPEORPLACE="SWIPENOTPLACE"
Mar  3 11:30:03 raspberrypi bash[863]: SECONDSWIPE="RESTART"
Mar  3 11:30:03 raspberrypi bash[863]: SECONDSWIPEPAUSE="2"
Mar  3 11:30:03 raspberrypi bash[863]: SECONDSWIPEPAUSECONTROLS="ON"
Mar  3 11:30:03 raspberrypi bash[863]: AUDIOIFACENAME="Headphone"
Mar  3 11:30:03 raspberrypi bash[863]: AUDIOIFACEACTIVE="0"
Mar  3 11:30:03 raspberrypi bash[863]: VOLUMEMANAGER="mpd"
Mar  3 11:30:03 raspberrypi bash[863]: AUDIOVOLCHANGESTEP="3"
Mar  3 11:30:03 raspberrypi bash[863]: AUDIOVOLMAXLIMIT="100"
Mar  3 11:30:03 raspberrypi bash[863]: AUDIOVOLMINLIMIT="1"
[... cut ...]
Mar  3 11:39:01 raspberrypi CRON[1366]: (root) CMD (  [ -x /usr/lib/php/sessionclean ] && if [ ! -d /run/systemd/system ]; then /usr/lib/php/sessionclean; fi)
Mar  3 11:39:01 raspberrypi systemd[1]: phpsessionclean.service: Succeeded.
Mar  3 11:39:01 raspberrypi systemd[1]: Started Clean php session files.
Mar  3 11:44:52 raspberrypi systemd[1]: Starting Cleanup of Temporary Directories...
Mar  3 11:44:52 raspberrypi systemd[1]: systemd-tmpfiles-clean.service: Succeeded.
Mar  3 11:44:52 raspberrypi systemd[1]: Started Cleanup of Temporary Directories.

dvalob, 2024-03-03:
just in case this can help, I used the options 1) and then 1 in the last configuration steps (I was a bit confused and wonder if I should have used the option 0 "RFID reader RFID Reader keyboard"?

Have you connected your RFID reader? [Y/n] y
Please select the RFID reader you want to use
1) USB-Reader (e.g. Neuftech)
2) RC522
3) PN532
4) Manual configuration
5) Multiple RFID reader
#? Choose the reader from list
0 RFID Reader RFID Reader Keyboard
1 RFID Reader RFID Reader
2 Dell Dell USB Keyboard

s-martin, 2024-03-03:
You could try to use the keyboard entry.

Run `python3 ./scripts/RegisterDevice.py` to register the reader again.

s-martin, 2024-03-03:
Just for clarification, did you swiped a card before retrieving the logs above?

dvalob, 2024-03-03:
> You could try to use the keyboard entry.
> 
> Run `python3 ./scripts/RegisterDevice.py` to register the reader again.

This worked after choosing keyboard and a reboot!!!! :) :) :) My daughter thanks you soooo much!!! :) :)

s-martin, 2024-03-03:
Glad it worked!