MiczFlor/RPi-Jukebox-RFID #2286: Ignoring failed package load for 'gpio.gpioz.plugin' on future3

Thilo-90, 2024-03-03:
### Describe your problem 

When setting up a new future3 installation i cannot use Buttons with the GPIO Pins.
I followed the installation guide and went through the installation process without any issues.

For ease of use i change the settings files in editor on windows with via samba.

I changed the `shared/settings/jukebox.yaml` file to
```
gpioz:
  enable: true
```
 and added an `gpio.yaml` with contents
```
# Provides a few selected examples of GPIO configuration
# Check out the documentation for many more configuration recipes
# documentation/builders/gpio.md
pin_factory:
  type: rpigpio.RPiGPIOFactory
output_devices:

input_devices:
  TogglePlayback:
    # A button to toggle music playback on single press
    type: Button
    kwargs:
      pin: 27
    actions:
      on_press:
        alias: toggle
```

I know the wiring is correct, as the box is working with my former version 2 Phoniebox. 
There seems to be a problem with the `gpio.gpioz.plugin` which i cannot solve on my own.

#### What's your hardware set up? 

RaspberryPi 4, 32bit OS Lite, 

#### If possible, try to attach logs from ... (paths from RPi)
Full error Log:
```
03.03.2024 12:44:01 -  588:plugs.py           - jb.plugin            - MainThread      - ERROR    - Ignoring failed package load for 'gpio.gpioz.plugin'
03.03.2024 12:44:01 -  589:plugs.py           - jb.plugin            - MainThread      - ERROR    - Reason: AttributeError: 'NoneType' object has no attribute 'items'
03.03.2024 12:44:01 -  590:plugs.py           - jb.plugin            - MainThread      - ERROR    - Detailed reason:
Traceback (most recent call last):
  File "/home/thilo/RPi-Jukebox-RFID/src/jukebox/jukebox/plugs.py", line 585, in load_all_named
    load(package, load_as, prefix)
  File "/home/thilo/RPi-Jukebox-RFID/src/jukebox/jukebox/plugs.py", line 575, in load
    raise e
  File "/home/thilo/RPi-Jukebox-RFID/src/jukebox/jukebox/plugs.py", line 572, in load
    func()
  File "/home/thilo/RPi-Jukebox-RFID/src/jukebox/components/gpio/gpioz/plugin/__init__.py", line 349, in initialize
    _build_all_output_devices()
  File "/home/thilo/RPi-Jukebox-RFID/src/jukebox/components/gpio/gpioz/plugin/__init__.py", line 183, in _build_all_output_devices
    for name, config in cfg_gpio.getn('output_devices', default={}).items():
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'items'

03.03.2024 12:44:01 -  146:daemon.py          - jb.daemon            - MainThread      - ERROR    - Plugins with errors during load: gpio
```

s-martin, 2024-03-03:
Could you try to remove the line `output_devices:` in gpio.yaml?

You didn't specify any output device anyway.

Thilo-90, 2024-03-03:
@s-martin 
tried it and the error is gone.
Though the Button is wired to GPIO27 (physical Pin 13) and is still not working.

s-martin, 2024-03-03:
> tried it and the error is gone.

🎉

> Though the Button is wired to GPIO27 (physical Pin 13) and is still not working.
> 
> When i press the button, the audio is not paused

Can you provide the log after you pressed the button?

Thilo-90, 2024-03-03:
which log? in the `app.log ` there is no entry if i press the button.

Thilo-90, 2024-03-03:
@s-martin nevermind... stupid mistake. now it is working after your fix.
The problem was of physical nature. with all the sd card switching, the round robin GDN wire broke...
I checked functionality with the old sd card running version 2 and also had no button presses. 
Temporary fix (tape) and everything works as intended.

AlvinSchiller, 2024-03-13:
@Thilo-90 anything left or can we close this?

Thilo-90, 2024-03-15:
@AlvinSchiller can be closed. Thanks for the help :)