MiczFlor/RPi-Jukebox-RFID #2290: 🐛 | no input possibility in search field

Lippsson, 2024-03-08:
### Version

3.5.2

### Branch

future3/main

### OS

Raspbian GNU/Linux 11 (bullseye)

### Pi model

3B+

### What happened?

[image]
I cant add any text into serch field

s-martin, 2024-03-08:
Could you add some logs after you tried to add some to the search field?

See here https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/future3/main/documentation/builders/troubleshooting.md#the-short-answer

Lippsson, 2024-03-08:
[errors.log](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/14543316/errors.log)
[app.log](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/14543317/app.log)

Sure.. Let me know if you need more information. Did it work for you?

pabera, 2024-03-10:
This is on me. I know about the issue and it should be fixed! :-)

Lippsson, 2024-03-10:
Great news! Can I update a file to make it to work, or do I need a complete reinstall?

pabera, 2024-03-16:
Fixed

pabera, 2024-03-16:
> Great news! Can I update a file to make it to work, or do I need a complete reinstall?

Should be possible. Will follow up with a path forward.

closed 2024-03-16 by pabera

pabera, 2024-03-19:
> Great news! Can I update a file to make it to work, or do I need a complete reinstall?

Here is a path forward -> https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/2300/files

PR #2292 (2024-03-16): fix: search filter in library
Fixes #2290