MiczFlor/RPi-Jukebox-RFID #2291: Startup/Shutdown sounds do not play 🐛 |

dvalob, 2024-03-15:
### Version

2.5.0

### Branch

master

### OS

Raspbian GNU/Linux 10 (buster)

### Pi model

3B 1.2

### Hardware

USB reader

### What happened?

Startup and shutdown sounds do not work. I have a segmentation fault in my logs. A similar issue was reported by the past but closed because too old (https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/1178).

My startup volume is correctly set to 25 (issue fixed in a previous release)

Mar  8 20:43:56 raspberrypi bash[913]: /home/pi/RPi-Jukebox-RFID/scripts/startup-scripts.sh: line 53:  1032 **Segmentation fault**      /usr/bin/mpg123 -f -${mpgvolume} /home/pi/RPi-Jukebox-RFID/shared/startupsound.mp3
Mar  8 20:43:56 raspberrypi bash[913]: Updating DB (#1) ...
Mar  8 20:43:56 raspberrypi bash[913]: volume: 25%   repeat: off   random: off   single: off   consume: off_

### Logs

ar  8 20:43:42 raspberrypi fake-hwclock[116]: Fri  8 Mar 20:43:38 UTC 2024
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] Booting Linux on physical CPU 0x0
Mar  8 20:43:42 raspberrypi systemd-fsck[135]: e2fsck 1.44.5 (15-Dec-2018)
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] Linux version 5.10.103-v7+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1529 SMP Tue Mar 8 12:21:37 GMT 2022
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] CPU: div instructions available: patching division code
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] random: fast init done
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] Memory policy: Data cache writealloc
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] Reserved memory: created CMA memory pool at 0x37400000, size 64 MiB
Mar  8 20:43:42 raspberrypi systemd[1]: Started udev Coldplug all Devices.
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] Zone ranges:
Mar  8 20:43:42 raspberrypi kernel: [    0.000000]   DMA      [mem 0x0000000000000000-0x000000003b3fffff]
Mar  8 20:43:42 raspberrypi kernel: [    0.000000]   Normal   empty
Mar  8 20:43:42 raspberrypi systemd[1]: Starting Helper to synchronize boot up for ifupdown...
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] Movable zone start for each node
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] Early memory node ranges
Mar  8 20:43:42 raspberrypi kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000003b3fffff]
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] On node 0 totalpages: 242688
Mar  8 20:43:42 raspberrypi kernel: [    0.000000]   DMA zone: 2133 pages used for memmap
Mar  8 20:43:42 raspberrypi systemd[1]: Started Helper to synchronize boot up for ifupdown.
Mar  8 20:43:42 raspberrypi kernel: [    0.000000]   DMA zone: 0 pages reserved
Mar  8 20:43:42 raspberrypi kernel: [    0.000000]   DMA zone: 242688 pages, LIFO batch:63
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] percpu: Embedded 20 pages/cpu s50828 r8192 d22900 u81920
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] pcpu-alloc: s50828 r8192 d22900 u81920 alloc=20*4096
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240555
Mar  8 20:43:42 raspberrypi systemd-fsck[135]: rootfs: clean, 114270/1797408 files, 943027/7748608 blocks
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=2ad10d96-02 rootfstype=ext4 fsck.repair=yes rootwait
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Mar  8 20:43:42 raspberrypi systemd[1]: Started File System Check on Root Device.
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] Memory: 878804K/970752K available (10240K kernel code, 1312K rwdata, 2952K rodata, 1024K init, 862K bss, 26412K reserved, 65536K cma-reserved)
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[... 591 lines cut ...]
Mar  8 20:43:57 raspberrypi mpd: update: updating Cuentos/Audiocuento El tigre y el conejo Cuento infantil (192kbit_AAC).m4a
Mar  8 20:43:57 raspberrypi mpd: update: updating Cuentos/¿Quién le pone el cascabel al gato_ _ Audiocuento con valores y sabiduría (192kbit_AAC).m4a
Mar  8 20:43:57 raspberrypi mpd: update: updating Cuentos/Antiguo consejo chino _ Cuento para reflexionar (192kbit_AAC).m4a
Mar  8 20:43:57 raspberrypi mpd: update: updating RocknRoll/Little Bitty Pretty One-Thurston Harris-1957 (96kbit_AAC).m4a
Mar  8 20:43:57 raspberrypi mpd: update: updating RocknRoll/Jerry Lee Lewis - Great Balls of Fire (128kbit_AAC).m4a
Mar  8 20:43:57 raspberrypi mpd: update: updating RocknRoll/Rock Around the Clock-Bill Haley-original song-1955 (128kbit_AAC).m4a
Mar  8 20:43:57 raspberrypi mpd: update: updating RocknRoll/The Crystals - He's a Rebel (1962) (128kbit_AAC).m4a
Mar  8 20:43:57 raspberrypi mpd: update: updating RocknRoll/Little Darlin'-The Diamonds-1957 (96kbit_AAC).m4a
Mar  8 20:43:57 raspberrypi mpd: update: updating RocknRoll/The Best Rock and Roll Songs of the 50s - part two (128kbit_AAC).m4a
Mar  8 20:43:58 raspberrypi mpd: update: updating RocknRoll/The Best Rock and Roll Songs of the 50s - part one (128kbit_AAC).m4a
Mar  8 20:43:58 raspberrypi mpd: exception: Failed to bind to '[::1]:6600'
Mar  8 20:43:58 raspberrypi mpd: exception: nested: Failed to bind socket: Address already in use
Mar  8 20:43:58 raspberrypi systemd[954]: mpd.service: Main process exited, code=exited, status=1/FAILURE
Mar  8 20:43:58 raspberrypi systemd[954]: mpd.service: Failed with result 'exit-code'.
Mar  8 20:43:58 raspberrypi systemd[954]: Failed to start Music Player Daemon.
Mar  8 20:43:58 raspberrypi systemd[954]: Reached target Default.
Mar  8 20:43:58 raspberrypi systemd[954]: Startup finished in 2.686s.
Mar  8 20:43:58 raspberrypi mpd: update: updating RocknRoll/Will You Still Love Me Tomorrow - The Shirelles (152kbit_Opus).opus
Mar  8 20:44:04 raspberrypi mpd: update: updating Modern_jazz/All About That Bass - Postmodern Jukebox European Tour Version (128kbit_AAC).m4a

Mar  8 20:44:10 raspberrypi systemd[1]: systemd-fsckd.service: Succeeded.
Mar  8 20:44:38 raspberrypi systemd-timesyncd[345]: Synchronized to time server for the first time [2001:418:3ff::53]:123 (2.debian.pool.ntp.org).
Mar  8 20:44:39 raspberrypi systemd[1]: systemd-hostnamed.service: Succeeded.
Mar  8 20:45:16 raspberrypi systemd[1]: phoniebox-idle-watchdog.service: Succeeded.
Mar  8 20:50:40 raspberrypi systemd[1]: Started Session 4 of user pi.

s-martin, 2024-03-15:
`/home/pi/RPi-Jukebox-RFID/scripts/startup-scripts.sh: line 53:  1032 Segmentation fault      /usr/bin/mpg123 -f **-${mpgvolume}** /home/pi/RPi-Jukebox-RFID/shared/startupsound.mp3`

I assume the not expanded variable `{mpgvolume}` is the issue.
Not sure what causes this, though.

dvalob, 2024-03-15:
it is declared here, right?
[image]

dvalob, 2024-03-15:
I do not think it will help. I have another phonie box where the start/stop sound play, and the code is exactlly the same on both:
[image]

They have the exactly same OS version, same HW. The left one is 2.4 - abcf6e92 - master, the right one is 2.5.0 - 4101984b - master

AlvinSchiller, 2024-03-15:
Do you use the audiojack or another audio output (like hifiberry)?

dvalob, 2024-03-15:
Jack output

AlvinSchiller, 2024-03-15:
What happens if you trigger the command manually?
`/usr/bin/mpg123 /home/pi/RPi-Jukebox-RFID/shared/startupsound.mp3`

dvalob, 2024-03-24:
Hi! Please find the output

dvalob, 2024-03-25:
[image]

And yes, I do have the file /home/pi/.asoundrc
[image]
[image]

dvalob, 2024-03-25:
no it was already there. I did what you suggested and now I can play both startup and shutdown sounds through /usr/bin/mpg123 /home/pi/RPi-Jukebox-RFID/shared/startupsound.mp3. Thanks!

When I start/stop the box I have both now. Thanks for your support!!

dvalob, 2024-03-25:
Should I close it myself, or let you close it (maybe with a note to fix it in the new versions?)

dvalob, 2024-03-25:
Closed when removing the .asoundrc file. thanks

closed 2024-03-25 by dvalob