MiczFlor/RPi-Jukebox-RFID #2302: 🐛 | docker: prebuild libzmq incompatible with debian-bullseye

notapirate, 2024-03-21:
### Version

3.6.0-alpha

### Branch

future3/develop

### OS

docker image debian:bullseye-slim

### Pi model

running in docker

### What happened?

The prebuilt libzmq package for x86_64 depends on glibc 2.34, but bullseye has 2.31 installed. This leads to an compile error when building pyzmq:

  /usr/bin/ld: /home/pi/RPi-Jukebox-RFID/.venv/lib/libzmq.so: undefined reference to `pthread_mutexattr_init@GLIBC_2.34'
  /usr/bin/ld: /home/pi/RPi-Jukebox-RFID/.venv/lib/libzmq.so: undefined reference to `__libc_single_threaded@GLIBC_2.32'
  /usr/bin/ld: /home/pi/RPi-Jukebox-RFID/.venv/lib/libzmq.so: undefined reference to `pthread_sigmask@GLIBC_2.32'
  /usr/bin/ld: /home/pi/RPi-Jukebox-RFID/.venv/lib/libzmq.so: undefined reference to `pthread_join@GLIBC_2.34'
  /usr/bin/ld: /home/pi/RPi-Jukebox-RFID/.venv/lib/libzmq.so: undefined reference to `pthread_setname_np@GLIBC_2.34'
  /usr/bin/ld: /home/pi/RPi-Jukebox-RFID/.venv/lib/libzmq.so: undefined reference to `pthread_mutex_trylock@GLIBC_2.34'
  /usr/bin/ld: /home/pi/RPi-Jukebox-RFID/.venv/lib/libzmq.so: undefined reference to `std::condition_variable::wait(std::unique_lock<std::mutex>&)@GLIBCXX_3.4.30'
  /usr/bin/ld: /home/pi/RPi-Jukebox-RFID/.venv/lib/libzmq.so: undefined reference to `std::__throw_bad_array_new_length()@GLIBCXX_3.4.29'
  /usr/bin/ld: /home/pi/RPi-Jukebox-RFID/.venv/lib/libzmq.so: undefined reference to `stat@GLIBC_2.33'
  /usr/bin/ld: /home/pi/RPi-Jukebox-RFID/.venv/lib/libzmq.so: undefined reference to `pthread_mutexattr_settype@GLIBC_2.34'
  /usr/bin/ld: /home/pi/RPi-Jukebox-RFID/.venv/lib/libzmq.so: undefined reference to `pthread_create@GLIBC_2.34'
  /usr/bin/ld: /home/pi/RPi-Jukebox-RFID/.venv/lib/libzmq.so: undefined reference to `pthread_cond_clockwait@GLIBC_2.34'
  /usr/bin/ld: /home/pi/RPi-Jukebox-RFID/.venv/lib/libzmq.so: undefined reference to `pthread_mutexattr_destroy@GLIBC_2.34'
  collect2: error: ld returned 1 exit status

  error: command '/usr/bin/x86_64-linux-gnu-gcc' failed with exit code 1

  Fatal: Falling back on bundled libzmq, but config has explicitly prohibited building the libzmq extension.

The bundled libzmq has no support for the DRAFTS api compiled, which causes runtime errors later.

Solution 1: Update to debian:bookworm which has glibc 2.36
Solution 2: @pabera builds the x86_64 variant of libzmq against glibc <=2.31
Solution 3: We build libzmq as part of the docker image creation

Note: I haven't checked all libzmq builds, but the armv7 build links against glibc 2.30 and is not affected by this problem

pabera, 2024-03-21:
Thanks @notapirate. It's what I thought. I like solution 2 most, and we had it in the past. It just took a lot longer than downloading it. But it remains the most painless solution!

pabera, 2024-04-08:
I believe this could be closed! Feel free to reopen in case something doesn't work.

closed 2024-04-08 by pabera