MiczFlor/RPi-Jukebox-RFID #2306: 🐛 | Unable to start playlists (web-app / rfid-reader)

FMuenke, 2024-03-24:
### Version

2.5.0 - 4101984b - master

### Branch

master

### OS

Raspbian GNU/Linux 11 (bullseye)

### Pi model

3B

### Hardware

RC522 RFID Reader
ALMO Speaker Set

### What happened?

I have just setup the raspi and installed the phonie box.

I am able to upload playlists and upload files into them, but when pressing the play-button in the playlist it states:

PLAYLIST is now playing, but it does not start, and it does not show up in the playlist (new songs list)

Finally, i can start the individual tracks in the playlist separately and as well add them manually to the next songs list.

I tried to reseach but i cannot find the solution.
Any ideas what i can try?

s-martin, 2024-03-24:
Did you check here?
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/MANUAL#adding-new-audio-files-and-create-a-playlist-for-a-new-card

FMuenke, 2024-03-25:
I was able to upload files through the web-app and i can see the files as well when i log into the rpi.
I can even start the tracks when i select them separately directly from the list of files below the folder in the web app)

When i hit the play button in the web-app of the folder it gives feedback: playlist is now playing.
But no response or audio further

AlvinSchiller, 2024-03-25:
Hi,
using playlists (e.g. m3u) files is not supported in version 2 as far as i know, and its not easliy added, as the current handling is folder based. It's confusing that the files are shown in the UI though.

Please have a look at version 3 where this feature has been added (https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/future3/main/documentation/builders/webapp/playlists-livestreams-podcasts.md#playlists).

FMuenke, 2024-03-25:
Thanks for the reply, i think i miss-used the word "playlist". I am referring to a folder with all files i want to play included.
I created a folder, i uploaded files into the folder (web-app) and tried to start the folder via web-app.
I cannot start the whole folder at once, but starting the individual tracks in the folder does work.

I hope a am not totally missing the point here.
But i confirmed the file are in the folder just pressing the play button of the folder does nothing.

Funnily i installed phniebox on my raspberry pi 2 and it workd perfectly there

AlvinSchiller, 2024-03-25:
Ok that's a different case :)
Can you supply you log file after trying to play the folder?

How many files are in this folder?

FMuenke, 2024-03-26:
In the folder there are 25 .m4a files.

in the `~/phoniebox_logs` folder i see the install file (i did rerun the install so there are two files)

In the `~/RPi-Jukebox-RFID/logs` there is only a _placeholeder_ file

here is the second install log-file:

`################################################
WIFIconfig="YES"
WIFIcountryCode="DE"
WIFIssid="FRITZ!Box 6490 Cable"
WIFIip="192.168.178.67"
WIFIipRouter="192.168.178.1"
EXISTINGuse=NO
AUTOHOTSPOTconfig="YES"
AUTOHOTSPOTssid="phoniebox"
AUTOHOTSPOTcountryCode="DE"
AUTOHOTSPOTpass="PlayItLoud"
AUTOHOTSPOTip="10.0.0.5"
AUDIOiFace="Headphone"
SPOTinstall="NO"
MPDconfig="NO"
DIRaudioFolders="/home/pi/RPi-Jukebox-RFID/shared/audiofolders"
GPIOconfig="YES"
WIFIconfig="YES"
WIFIcountryCode="DE"
WIFIssid="FRITZ!Box 6490 Cable"
WIFIip="192.168.178.67"
WIFIipRouter="192.168.178.1"
EXISTINGuse=NO
AUTOHOTSPOTconfig="YES"
AUTOHOTSPOTssid="phoniebox"
AUTOHOTSPOTcountryCode="DE"
AUTOHOTSPOTpass="PlayItLoud"
AUTOHOTSPOTip="10.0.0.5"
AUDIOiFace="Headphone"
SPOTinstall="NO"
MPDconfig="NO"
DIRaudioFolders="/home/pi/RPi-Jukebox-RFID/shared/audiofolders"
GPIOconfig="YES"
################################################
Generating locales (this might take a while)...
  en_GB.UTF-8... done
Generation complete.
Cloning into 'RPi-Jukebox-RFID'...
Writing to /root/.config/pip/pip.conf
Installing additional Python packages...
Configuring Samba...
Configuring web server...
already enabled
Run "service lighttpd force-reload" to enable changes
already enabled
Run "service lighttpd force-reload" to enable changes
### Deleting older versions of service daemons. This might throw errors, ignore them
Failed to disable unit: Unit file idle-watchdog.service does not exist.
Failed to disable unit: Unit file rfid-reader.service does not exist.
Failed to disable unit: Unit file phoniebox-startup-sound.service does not exist.
Failed to disable unit: Unit file gpio-buttons.service does not exist.
Failed to disable unit: Unit file phoniebox-rotary-encoder.service does not exist.
Failed to disable unit: Unit file phoniebox-gpio-buttons.service does not exist.
rm: cannot remove '/etc/systemd/system/rfid-reader.service': No such file or directory
rm: cannot remove '/etc/systemd/system/startup-sound.service': No such file or directory
rm: cannot remove '/etc/systemd/system/gpio-buttons.service': No such file or directory
rm: cannot remove '/etc/systemd/system/idle-watchdog.service': No such file or directory
rm: cannot remove '/etc/systemd/system/phoniebox-rotary-encoder.service': No such file or directory
rm: cannot remove '/etc/systemd/system/phoniebox-gpio-buttons.service': No such file or directory
### Done with erasing old daemons. Stop ignoring errors!
Updating DB (#2) ...
volume: n/a   repeat: off   random: off   single: off   consume: off
Setting /etc/dhcpcd.conf...
Setting /etc/wpa_supplicant/wpa_supplicant.conf...
Starting dhcpcd service...
Synchronizing state of hostapd.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install disable hostapd
Synchronizing state of dnsmasq.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install disable dnsmasq
Setting owner and permissions for directories...
numid=1,iface=MIXER,name='PCM Playback Volume'
  ; type=INTEGER,access=rw---R--,values=1,min=-10239,max=400,step=0
  : values=400
  | dBscale-min=-102.39dB,step=0.01dB,mute=1

#
# INSTALLATION FINISHED
#
#####################################################

If you are using an RFID reader, connect it to your RPi.
(In case your RFID reader required soldering, consult the manual.)
Have you connected your RFID reader? [Y/n] Y
Please select the RFID reader you want to use
1) USB-Reader (e.g. Neuftech)
2) RC522
3) PN532
4) Manual configuration
5) Multiple RFID reader
#? Please make sure that the RC522 reader is wired up correctly to the GPIO ports before continuing...
Continue (y/n)? Installing Python requirements for RC522...
Activating SPI...
Configure RFID reader in Phoniebox...
Restarting phoniebox-rfid-reader service...
Done.

DONE. Let the sounds begin.
Find more information and documentation on the github account:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/
Reboot is needed to activate all settings
Would you like to reboot now? [Y/n] Y`

FMuenke, 2024-03-26:
Ah i have found these logs after i have enabled WebApp-API logs:

`$responseList: {"repeat":"0","random":"0","single":"0","consume":"0","partition":"default","playlist":"7","playlistlength":"0","mixrampdb":"0.000000","state":"stop","chapters":[]}GET  ######### SCRIPT inc.readArgsFromCommandLine.sh (2024-03-26.11:23:15) ##
  # VAR CARDID: 
  # VAR COMMAND: savepos
  # VAR FOLDER: 
  # VAR VALUE: 
  ######### SCRIPT inc.readArgsFromCommandLine.sh (2024-03-26.11:23:15) ##
  # VAR CARDID: 
  # VAR COMMAND: playlistaddplay
  # VAR FOLDER: ilias
  # VAR VALUE: ilias
  ######### SCRIPT inc.readArgsFromCommandLine.sh (2024-03-26.11:23:15) ##
  # VAR CARDID: 
  # VAR COMMAND: single_check
  # VAR FOLDER: ilias
  # VAR VALUE: 
  ######### SCRIPT inc.readArgsFromCommandLine.sh (2024-03-26.11:23:15) ##
  # VAR CARDID: 
  # VAR COMMAND: shuffle_check
  # VAR FOLDER: ilias
  # VAR VALUE: 
  ######### SCRIPT inc.readArgsFromCommandLine.sh (2024-03-26.11:23:15) ##
  # VAR CARDID: 
  # VAR COMMAND: resume
  # VAR FOLDER: ilias
  # VAR VALUE: 

# WebApp API # /home/pi/RPi-Jukebox-RFID/htdocs/api/player.php
  # $_SERVER['REQUEST_METHOD']: GET
  # function handleGet() 
$responseList: {"repeat":"0","random":"0","single":"0","consume":"0","partition":"default","playlist":"8","playlistlength":"0","mixrampdb":"0.000000","state":"stop","chapters":[]}GET
# WebApp API # /home/pi/RPi-Jukebox-RFID/htdocs/api/playlist.php
  # $_SERVER['REQUEST_METHOD']: GET
  # function execSuccessfully: echo 'playlistinfo
close' | nc -w 1 localhost 6600
# WebApp API # /home/pi/RPi-Jukebox-RFID/htdocs/api/cover.php
# $_SERVER['REQUEST_METHOD']: GET
# WebApp API # /home/pi/RPi-Jukebox-RFID/htdocs/api/player.php
  # $_SERVER['REQUEST_METHOD']: GET
  # function handleGet() 
$responseList: {"repeat":"0","random":"0","single":"0","consume":"0","partition":"default","playlist":"8","playlistlength":"0","mixrampdb":"0.000000","state":"stop","chapters":[]}GET`

AlvinSchiller, 2024-03-26:
Can you also supply the output of `tail -n 200 /var/log/syslog`?

AlvinSchiller, 2024-03-26:
I read from your installation log that you chose `NO` at MPD configuration. For a specific reason?
Can you supply your `/etc/mpd.conf`?

s-martin, 2024-03-26:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/scripts%2Finstallscripts%2Finstall-jukebox.sh#L648-L650

Here is the question, if MPD should be configured.

From my point of view it doesn't make any sense, why the user should decide that, because Phoniebox is broken, if you decide NO.

Should be fixed.

s-martin, 2024-03-26:
Second issue:

In https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/develop/misc%2Fsampleconfigs

we have a `mpd.conf.sample` and a `mpd.conf.buster-default.sample`. this should be harmonized and checked, which one the install script uses.

AlvinSchiller, 2024-03-26:
> For the MPD-Config, it stated that i could also do it later and it was not necessary for my raspberry pi 2 (where is successfully installed it), so i did not consider it further. I really hope this is now not the issue

I believe that's the problem, as the path don't match.

I have look in the installation settings later and can supply a config ready to use.

Or you rerun the installation and select the option.

FMuenke, 2024-03-26:
Hi everyone, sorry for the ticket, it was indeed the MPD config.
I rerun the installation and this time configuring the MPD as it was stated in the installation command.

Now everything is working perfectly. Thanks for the patience, people like you make the internet work.

closed 2024-03-26 by FMuenke