MiczFlor/RPi-Jukebox-RFID #2313: 🐛 | bookworm/kernel 6.6 GPIO: Failed to add edge detection

varac, 2024-03-30:
### Version

3.5.2

### Branch

future/main

### OS

RPI OS bookworm lite - 32bit

### Pi model

3 B+

### Hardware

- [Suptronics x400](http://www.suptronics.com/xseries/x400.html) Soundcard
- [MFRC522 RFID Reader](https://rpi-jukebox-rfid.readthedocs.io/en/latest/rfid/mfrc522_spi.html)
- GPIO buttons

### What happened?

After a fresh install of RPI OS bookworm lite, 32-bit with latest `future/main`, the GPIO buttons and the RFID reader stopped working (They worked before with the same configuration on an older 2.x installation on RPI OS bullseye).
I saw [this discussion thread](https://github.com/MiczFlor/RPi-Jukebox-RFID/discussions/2295) which confirms this issue, but I wanted to create a dedicated issue for it, to not bloat the discussion topic too much since it covers also other issues of a bookworm/future/main installation. 

I can see this relevant log entry:

```
Exception in thread read_00Thread:
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/home/varac/RPi-Jukebox-RFID/src/jukebox/components/rfid/reader/__init__.py", line 143, in run
    self._reader = self._reader_module.ReaderClass(self._reader_cfg_key)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/varac/RPi-Jukebox-RFID/src/jukebox/components/rfid/hardware/rc522_spi/rc522_spi.py", line 85, in __init__
    self.device = pirc522.RFID(bus=spi_bus,
                  ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/varac/RPi-Jukebox-RFID/.venv/lib/python3.11/site-packages/pirc522/rfid.py", line 103, in __init__
    GPIO.add_event_detect(pin_irq, GPIO.FALLING,
RuntimeError: Failed to add edge detection
```

In the discussion [Sharing tips for future3 on bookworm/kernel 6.6 and upstreaming fixes #2295](https://github.com/MiczFlor/RPi-Jukebox-RFID/discussions/2295) @hoffie pointed out possible ways forward but I am missing concrete instructions know how to actually fix this issue on my current installation. Any help welcome, thanks already !

### Logs

[app.log](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/14812716/app.log)
[errors.log](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/14812717/errors.log)
[INSTALL-1710188086.log](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/14812718/INSTALL-1710188086.log)

### Configuration

[rfid.yaml.txt](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/14812730/rfid.yaml.txt)
[jukebox.yaml.txt](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/14812731/jukebox.yaml.txt)
[gpio.yaml.txt](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/14812733/gpio.yaml.txt)

AlvinSchiller, 2024-03-30:
Hi, currently there is no immediate solution, other then using bullseye.

The linked discussion suggests swapping the pirc lib, so you could try this by uninstalling the old one and install from the linked repo.

hoffie, 2024-04-01:
> I am missing concrete instructions know how to actually fix this issue on my current installation. 

Here is what I did:

**Use system lgpio (because the version on pypi is too old):**
(Update: Maybe this step is no longer necessary, since 0.2.2.0 seems to be on pypi since 2024-03-29?)
```
rm /home/phoniebox/RPi-Jukebox-RFID/.venv/lib/python3.11/site-packages/lgpio.py /home/phoniebox/RPi-Jukebox-RFID/.venv/lib/python3.11/site-packages/lgpio-0.2.2.0.egg-info /home/phoniebox/RPi-Jukebox-RFID/.venv/lib/python3.11/site-packages/_lgpio.cpython-311-arm-linux-gnueabihf.so
ln -s /usr/lib/python3/dist-packages/lgpio.py /home/phoniebox/RPi-Jukebox-RFID/.venv/lib/python3.11/site-packages/lgpio.py
ln -s /usr/lib/python3/dist-packages/lgpio-0.2.2.0.egg-info/ /home/phoniebox/RPi-Jukebox-RFID/.venv/lib/python3.11/site-packages/lgpio-0.2.2.0.egg-info
ln -s /usr/lib/python3/dist-packages/_lgpio.cpython-311-arm-linux-gnueabihf.so /home/phoniebox/RPi-Jukebox-RFID/.venv/lib/python3.11/site-packages/_lgpio.cpython-311-arm-linux-gnueabihf.so
```

**Configure Jukebox gpio.yaml to use LGPIO for buttons:**
I'm using this in my gpio.yml and it works (other Factories may work as well, I haven't tried):
```yaml
pin_factory:
  # Default is RPi.GPIO, which can't handle buttons on kernel 6.6+ anymore:
  type: lgpio.LGPIOFactory
```

**Replace PIRC522 by PIRC522-gpiozero:**
```
bash -c "source /home/phoniebox/RPi-Jukebox-RFID/.venv/bin/activate && pip uninstall pi-rc522 && pip install git+https://github.com/hoffie/pi-rc522-gpiozero/"`
```

You could also try to replace the broken RPi.GPIO by the following shim. I haven't tested it though. It might still need the above lgpio changes, but it should make the other changes redundant:
https://rpi-lgpio.readthedocs.io/en/release-0.4/install.html

varac, 2024-04-06:
@hoffie Thanks, I can confirm that your instructions work, my rfc 522 reader works again :tada: !
I don't know how to properly include this fix in the install script though, but it works as a reference workaround to make the rfc module working with current bookworm PiOS.

varac, 2024-04-06:
I wonder if it would make sense to create a `bookworm` issue label to gather all bookworm-related issues. I'd be happy to test if needed.

s-martin, 2024-04-27:
I tend to the lgpio and the rc522 gpiozero solution, because I think while the shim might be the most non-intrusive solution right now I'm not sure, if this will work as a long term solution.

Don't know what others think here?

@hoffie, did you make a PR to the rc522, so this fix may be included?
Nevertheless I think we could also use your branch.

hoffie, 2024-05-16:
> @hoffie, did you make a PR to the rc522, so this fix may be included?

Not yet, because the changes seemed a bit too fundamental. I'll try to find some time in the coming days.
(Sorry, I was rather unresponsive, also on my other interactions here; real life is rather busy right now)

s-martin, 2024-05-16:
Real life always comes first ;)

ben0r33, 2024-05-22:
So I can confirm @hoffie second solution by simple uninstalling rpi-gpio and install rpi-lgpio via pip to make rfid reader rc522 work again with latest linux kernel 6.6.28 on Raspbian Bookworm with RPI4.
```
systemctl --user stop jukebox-daemon
source .venv/bin/activate
pip uninstall rpi-gpio
pip install rpi-lgpio
systemctl --user start jukebox-daemon
```

Maybe the best option would be always using rpi-lgpio moving forward?

s-martin, 2024-05-22:
Right now I think using the shim `rpi-lgpio` looks like the easiest solution (despite my comments earlier), because it's a one line replacement in requirements.txt

Only question to @pabera or @AlvinSchiller: I would like to use a RPi.GPIO uninstall command to make it more robust, where would be a good place in the installation scripts?

AlvinSchiller, 2024-05-24:
> Right now I think using the shim `rpi-lgpio` looks like the easiest solution (despite my comments earlier), because it's a one line replacement in requirements.txt
>
> Only question to @pabera or [...]

For v2 this could be done right before the apt installation. 
For v3 we have currently no automated update path, so this would be only a hint in the documentation / release notes (like we already have in the [current develop version](https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/2258#issuecomment-2054198210) )

alesrosina, 2024-06-07:
Hello, I'm now trying to install it on a bookworm, and rfid works after I've replaced `rpio-gpio` with `rpi-lgpio` (according to  this comment https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/2313#issuecomment-2124397117). I'm using rc522 rfid reader.

With two issues that I'm not sure why it's happening:
- gpio buttons (rotary button, led light and a normal button) does not seems to work. I cannot see any errors anywhere, and since rfid is working, not sure what's going on
- rfid does not read Ntag215 stikers (https://www.berrybase.de/en/rfid/nfc-tags-ntag215-25mm-selbstklebend-weiss-10-stueck), but does read Mifare Classic1K RFID. Again, no errors that I can find in logs

Same setup worked on raspberry 1 with previous version of raspberry os (so all gpio buttons & rfid reader), but now I've changed it with raspberry 3B+ and installed also 64 bit OS. I know you're mentioning, that we should use 32 bit version, but since all other things are working on 64 bit os, I don't see why this should not. For installation on 64 bit i removed checks in install script and it (mostly) worked. 

Any ideas would be appreciated.

s-martin, 2024-06-07:
> With two issues that I'm not sure why it's happening:
> - gpio buttons (rotary button, led light and a normal button) does not seems to work. I cannot see any errors anywhere, and since rfid is working, not sure what's going on

Can you post your gpio.yaml and the logs?

> - rfid does not read Ntag215 stikers (https://www.berrybase.de/en/rfid/nfc-tags-ntag215-25mm-selbstklebend-weiss-10-stueck), but does read Mifare Classic1K RFID. Again, no errors that I can find in logs

Ntags don't work at the moment. See #2373

AlvinSchiller, 2024-06-07:
> Same setup worked on raspberry 1 with previous version of raspberry os

Are you sure that the NTAGs also worked with the previous setup? That wouldn't match our observation.

alesrosina, 2024-06-07:
> > Same setup worked on raspberry 1 with previous version of raspberry os
> 
> Are you sure that the NTAGs also worked with the previous setup? That wouldn't match our observation.

Now I'm not sure anymore :)

It definitely worked on v2, but when I've installed v3 I'm no longer sure, since on raspi 1 it never worked nicely (web interface didnt respond well) and I assumed, this was due to lack of power on raspi1. So afterwards it was never fully in use, until I've now stated playing around with raspi3.

alesrosina, 2024-06-07:
> Can you post your gpio.yaml (assuming you use V3) and the logs?

Sure:

```
pin_factory:
  type: lgpio.LGPIOFactory

input_devices:
  VolumeRotator:
    type: RotaryEncoder
    kwargs:
      a: 27
      b: 22
    actions:
      on_rotate_clockwise:
        alias: change_volume
        args: 5
      on_rotate_counter_clockwise:
        alias: change_volume
        args: -5

output_devices:
  StatusLED:
    type: LED
    connect:
      - gpio.gpioz.plugin.connectivity.register_status_led_callback
    kwargs:
      pin: 14
[app.log](https://github.com/user-attachments/files/15728795/app.log)

```
Here's also app.log:

[app.log](https://github.com/user-attachments/files/15728795/app.log)

AlvinSchiller, 2024-06-07:
> pin_factory: type: lgpio.LGPIOFactory 

I think the changed factory is not needed if you only swapped the libs. Can you try with the "old"?

alesrosina, 2024-06-07:
> > pin_factory: type: lgpio.LGPIOFactory
> 
> I think the changed factory is not needed if you only swapped the libs. Can you try with the "old"/original?

Tried. Same result.

Should I also try to replace libs as suggested in first comment? https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/2313#issuecomment-2030579144 
Probably not important, since rfid reader works ...

s-martin, 2024-06-07:
> 
> Should I also try to replace libs as suggested in first comment? https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/2313#issuecomment-2030579144 
> Probably not important, since rfid reader works ...

No, as your reader works....

alesrosina, 2024-06-07:
> > Should I also try to replace libs as suggested in first comment? [#2313 (comment)](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/2313#issuecomment-2030579144)
> > Probably not important, since rfid reader works ...
> 
> No, as your reader works....

Is there anything else I could provide, just to get some ideas what's going on? Or any direction in which I could be looking.

One thing I forgot to metion is, that I've did an install directly from `future3/develop` branch.

s-martin, 2024-06-07:
Did you activate GPIO in jukebox.yaml?

Line 106:
```
gpioz:
  enable: true
  config_file: ../../shared/settings/gpio.yaml
```

`enable` needs to be `true`

alesrosina, 2024-06-07:
> Did you activate GPIO in jukebox.yaml?
> 
> Line 106:
> 
> ```
> gpioz:
>   enable: true
>   config_file: ../../shared/settings/gpio.yaml
> ```
> 
> `enable` needs to be `true`

Oh. :see_no_evil: Of course, this was the reason. Thanks for this, this was a very obvious mistake.

So with this out of the way, I can confirm v3 is working as it should on raspberian 64bit bookworm (raspberry 3B+), so I guess this limitation could be removed from setup, right?

s-martin, 2024-06-07:
Cool!

Thanks for trying 64 bit Bookworm. So we will figure out how to handle 64bit in the future.

plimptm, 2024-06-27:
I just took another stab at this on a pi zero w (armv6l) and the `pip install rpi-lgpio` failed as unable to to build the wheel for lgpio. I tested in a clean venv using the `--system-site-packages` option (per [this rpi-lgpio doc page](https://github.com/waveform80/rpi-lgpio/blob/main/docs/install.rst#in-venv-with-system-packages)) and that met the lgpio dependency for the shim to install. I believe that means the symlink strategy @hoffie suggested [here](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/2313#issuecomment-2030579144) would also work, though using the rpi-lgpio shim might render the rest of those steps unecessary.

rgbetanco, 2024-08-14:
I am using Debian GNU/Linux 12 (bookworm) on Raspberry Pi 5, I had cloned the branch pi-rc522-gpiozero and ran the setup.py in it. I have uninstall rpi-gpio and install rpi-lgpio. I connected the RC522 module to the pi and ran the example code. Nothing happens, no card is detected. Pi5 is 64bits.

alexxx-k, 2024-10-31:
Hello,

i stumbled accross the issue as well on my RP4 running 32bit 
Linux phoniebox 6.6.51+rpt-rpi-v7l #1 SMP Raspbian 1:6.6.51-1+rpt3 (2024-10-08) armv7l GNU/Linux.

First i thought my wiring is wrong, so i followed a tiny tutorial (https://www.youtube.com/watch?v=evRuZRxvPFI)

from there i tested these two scripts.
This does work very well. Maybe this is a possible solution as well.

[example.zip](https://github.com/user-attachments/files/17591726/example.zip)

alexxx-k, 2024-10-31:
Hello, I made a testfile that get my reader working.
This file breaks the _read_card_legacy method. I basically do not understand what its purpose so i'm cannot make any fix here.

RPi-Jukebox-RFID/src/jukebox/components/rfid/hardware/rc522_spi/rc522_spi.py
[rc522_spi.zip](https://github.com/user-attachments/files/17593146/rc522_spi.zip)

michael-dev, 2024-12-21:
For v2 with bookworm and a fresh install, I ended up with to rpi-lgpio versions:
/usr/local/lib/python3.11/dist-packages/RPi/GPIO
/usr/lib/python3/dist-packages/RPi/GPIO

The one in the local path was outdated, resulting in the PhonieboxGPIOControl[4154]: RuntimeError: Failed to add edge detection" errors. In removed it, and the box now works fine.

closed 2025-01-20 by s-martin