MiczFlor/RPi-Jukebox-RFID #2317: 🚀 | Play a specific song on repeat

BoldBigflank, 2024-04-03:
### Feature

When I tap a card, I would like to play a 60 second white noise file on repeat indefinitely.

### User perspective

The simplest way I imagine could be to have an option tied to the "Play music" action.

A more thorough approach would be to allow a card action to be to implement multiple actions on trigger and also set actions for specific events such as action ended or left action. Then, I could set the trigger to both execute the "Audio & Volume > Repeat > Enable Repeat Song" action as well as the "Play Music" action, and then when the card is changed, execute "Audio & Volume > Repeat > Disable" action.

pabera, 2024-04-15:
Thanks for the request. It should be possible to apply multiple RPC calls to a single card assignments.

For this, we need to update the `cards.yml` structure to accept these multiple calls. At the moment, they only support a single use case

Current structure to play an album

```yaml
'321234568':
  alias: play_album
  args:
  - Bibi & Tina
  - Jetzt in Echt - Soundtrack zum Kinofilm
```

Future structure to support `n` calls, in this example Play an Album on Repeat and a specific volume.

```yaml
'321234568':
  actions:
    - alias: play_album
      args:
        albumartist: Bibi & Tina
        album: Jetzt in Echt - Soundtrack zum Kinofilm
    - alias: repeat
      args:
        option: toggle_repeat
    - alias: set_volume
      args:
        volume: 60
```

A single action would look like this

```yaml
'321234568':
  actions:
    - alias: play_album
      args:
        albumartist: Bibi & Tina
        album: Jetzt in Echt - Soundtrack zum Kinofilm
```

The code would have to be changed to run through a loop of actions and call all RPC commands after each other.

Things to consider

1. All previous `cards.yml` must be rewritten to the new structure
2. Update UI to manage multiple commands
3. When created manually, try to avoid running same commands within the same card assignments.

If you have further comments, please add as comments.

s-martin, 2024-04-16:
I like the approach.

If we do this we also should provide a feature to automatically convert cards.yaml to the new format as the cards are fundamental to Phoniebox.

This would also add to a possibility to update not only reinstall.

plimptm, 2024-06-28:
I'm wondering if it would be worth considering a different way of implementing this. Here's another idea: create a new top-level RPC wrapper function/alias (e.g. `sequence`) where the input args would be the array of actions each with their own args.

Example: 
```yaml
'321234568':
  alias: sequence
  args:
    - alias: play_album
      args:
        albumartist: Bibi & Tina
        album: Jetzt in Echt - Soundtrack zum Kinofilm
    - alias: repeat
      args:
        option: toggle_repeat
    - alias: set_volume
      args:
        volume: 60
```

This approach would address the first consideration @pabera mentioned above and provide backwards compatibility with simple single action card entries.

I'm honestly not sure if an RPC call can handle data structures as input arguments or if they would need to be serialized into a string and parsed on the other end in order to issue the 'real commands'.

Just offering the idea for consideration since it doesn't appear this enhancement is underway yet.