MiczFlor/RPi-Jukebox-RFID #2327: 🚀 | "next_song" performing no action when no next song is available

IgorWalton, 2024-04-08:
### Feature

I would like the next_song GPIO button to do nothing, when there is no next song available (last track of a folder or single track without any folder). Currently this will stop the playback.

### User perspective

I imagine my (very young) child will pretty randomly press all of the available buttons, so this behaviour would be better suited.

### Further information

using Jukebox Core Version 3.5.2

Lippsson, 2024-04-08:
Good feature request. It would be great if that above is the standard behavior

pabera, 2024-04-08:
I think this is what the following PR is trying to solve?!

* #2326

hoffie, 2024-04-11:
Hrm, the linked PR will behave in the following way:
1. When the last song of a playlist is being played and next() is called, the default will be to stop playing (this is configurable).
1. When the player is stopped (e.g. because the last song of a playlist finished) and the next() is called, the player will rewind the playlist (i.e. it will start playing the first song of the playlist again).

@IgorWalton Is that what you're after? If not, we would have to make (2) configurable as well.

IgorWalton, 2024-04-11:
1. I personally would prefer if the default is "no action" in this situation
2. The way I think would make the most sense is during a stop in the middle of a track "next" jumps to the next track in the current playlist, and at the end of a playlist it there's either "no action" or "start playing the first song of the playlist"
I don't have the developer perspective (yet), so just 2 Cents from a User's perspective...

hoffie, 2024-04-11:
As #2326 seems to be merged soonish, I can prepare another PR to add the *no action* behavior for (1).
Regarding (2), this can certainly be implemented as well. I was a bit hesitant to make this configurable, because every option adds code and config complexity, but I guess it's worth it if there are people who want to use it like that. :)

s-martin, 2024-04-11:
You can still add it, if you want

hoffie, 2024-04-11:
> I personally would prefer if the default is "no action" in this situation

#2326 should now technically cover this.

> The way I think would make the most sense is during a stop in the middle of a track "next" jumps to the next track in the current playlist

Done.

> at the end of a playlist it there's either "no action" or "start playing the first song of the playlist"

This should now behave similar to the case when running `next()` while playling the last song of the playlist (= it's configurable).

s-martin, 2024-04-12:
#2326 has been merged to future3/develop

PR #2326 (2024-04-12): Fix PlayerMPD.prev/next() when stopped
Fix PlayerMPD.prev/next() when stopped

* Avoid MPD-related crashes during all prev/next() calls.
* Explicitly handle prev() in stopped state, configurable via 
  `playermpd.stopped_prev_action`.
* Explicitly handle next() in stopped state, configurable via 
  `playermpd.stopped_next_action`.
* Explicitly handle next() when reaching the end of the playlist:
  jukebox-daemon will now ignore the action by default (similar to v2).
  It can also be configured to rewind the playlist instead by setting
  the new config option `playermpd.end_of_playlist_next_act [... cut]
commit a9ab571a (2024-04-12): Fix PlayerMPD.prev/next() when stopped (#2326)

* utils: Add get_config_action

This abstracts away the functionality to resolve a given config option
to an action in a pre-defined dict.

* Fix PlayerMPD.prev/next() when stopped

* Avoid MPD-related crashes during all prev/next() calls.
* Explicitly handle prev() in stopped state, configurable via
  `playermpd.stopped_prev_action`.
* Explicitly handle next() in stopped state, configurable via
  `playermpd.stopped_next_action`.
* Explicitly handle next() when reaching the end of the playlist:
  jukebox-daemon will now ignore the action by defau [... cut]
closed 2024-04-12 by s-martin