MiczFlor/RPi-Jukebox-RFID #2329: 🐛 | 3.6 crash when change to album view

Lippsson, 2024-04-10:
### Version

3.6.0 alpha

### Branch

future3 develop

### OS

Bookworm v12 32bit

### Pi model

Pi4 2GB

### What happened?

When changing in Browser from Folder view to Album its crashing: 
[image]

pabera, 2024-04-10:
This is most likely a result of a Jukebox crash. If you could figure out the logs from the Jukebox daemon, it'll be easier to dive into it :-)

Lippsson, 2024-04-10:
[errors.log.zip](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/14936911/errors.log.zip)
Does this logfiles helps you?

pabera, 2024-04-10:
Yo, this helps. It's related to the CoverArtCache. It seems to have trouble reading specific files, like `m4a`, or `flac`

Would you mind joining our [chat](https://matrix.to/#/#phoniebox_community:matrix.org). I found a few interesting problems in your file which I like to chat about.

Lippsson, 2024-04-23:
I have tested the new pull request and here is my feedback. https://youtu.be/lRqHZwdUcfQ
I also shared already the link to @pabera and he will take a look into it when he have time.

pabera, 2024-04-23:
According to the video, there are still some hick-ups when toggling between Album/Folder view with CoverArt enabled. This needs to be investigated.