MiczFlor/RPi-Jukebox-RFID #2354: 🚀 | Default settings.yaml

pabera, 2024-04-22:
### Feature

I recommend introducing a `settings.yaml` file for constants that users (builders) don't need to modify. Currently, our `jukebox.yaml` combines both user-relevant settings and constants. One might wonder why these constants aren't simply coded into the software, but occasionally, even these constants need adjustments, such as during development.

In our existing `jukebox.default.yaml`, there are many details that aren't relevant to the builder, and altering them could impair the system's functionality. 

My suggestion involves adopting an approach similar to Docker Compose yaml files, where `settings.yaml` serves as the base and `jukebox.yaml` acts as an override.

Here are several sections that could be separated out (as the comments also advise against modifying).

``` yaml
modules:
  named:
    # Do not change the order!
    publishing: publishing
    volume: volume
    jingle: jingle
    jingle.alsawave: jingle.alsawave
    jingle.jinglemp3: jingle.jinglemp3
    player: playermpd
    cards: rfid.cards
    rfid: rfid.reader
    timers: timers
    host: hostif.linux
    bluetooth_audio_buttons: controls.bluetooth_audio_buttons
    gpio: gpio.gpioz.plugin
    sync_rfidcards: synchronisation.rfidcards
  others:
  - misc
```

``` yaml
rpc:
  tcp_port: 5555
  websocket_port: 5556
publishing:
  # Ports for the publisher. Setting a port number to 0 disables the port
  # The WebUI requires the websocket connection
  tcp_port: 5558
  websocket_port: 5557
rfid:
  reader_config: ../../shared/settings/rfid.yaml
  card_database: ../../shared/settings/cards.yaml
```

``` yaml
webapp:
  coverart_cache_path: ../../src/webapp/build/cover-cache
```

s-martin, 2024-04-22:
I think it's a good idea.
Only suggestion for clarity: call the file with the non changing parameters `constants.yaml`, `base.yaml` or `default.yaml`.

I think `settings` sounds like the users should change something here.

s-martin, 2024-04-25:
I think those could also be moved to default settings:

```
playermpd:
  status_file: ../../shared/settings/music_player_status.json
  mpd_conf: ~/.config/mpd/mpd.conf
```

is it possible that some parameters of a section (e.g. `playermpd`) are in the default file and some are the file, which could/should be changed.