MiczFlor/RPi-Jukebox-RFID #2364: Support (multiple) Displays

s-martin, 2024-05-04:
### Feature

Phoniebox V3 should support displays (e.g. [Pirate Audio HATs](https://shop.pimoroni.com/collections/pirate-audio) or [HD44780](https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/develop/components/displays/HD44780-i2c).

### User perspective

Displays should show information about currently played songs (e.g. album, title, etc), maybe even cover art.

s-martin, 2024-05-04:
#1112 would be replaced by a new implementation

Lippsson, 2024-05-06:
I'm currently testing whether I can rewrite a script for the small 1.3" displays (128x64 pixels). The[ script from Splitti ](https://github.com/splitti/oled_phoniebox)still works, but I'm missing things like the battery symbol and the indicators for shuffle and repeat. 
If anyone has the time, desire and knowledge, help is welcome and appreciated.
 
[image]

s-martin, 2024-05-07:
👍
I don't know how much the script from splitti relies on the Phoniebox implementation (V2).

This issue should target the V3 implementation.

s-martin, 2024-05-07:
I envision the implementation in two parts:

# Access/control of the display hardware ("backend", if you will)

This should be specific to the used display hardware, but every backend should provide a `show()` method, which is called by the frontend to show stuff on the display.

In addition methods like `reload`, `clear`, etc.

# Generic, configurable display ("frontend")

The user should be able to configure the information to be displayed, e.g. artist, song name, cover art, maybe meta info (such as battery status or Phoniebox version, etc.).

This is of course dependent of the used backend/hardware, e.g. resolution etc.

This part should handle the RPC  to retrieve information to be displayed.

s-martin, 2024-05-07:
#1710

s-martin, 2024-05-10:
- Library for LCD: PirateAudio? HD44780? https://github.com/rm-hull/luma.lcd
- Library for LED Matrix: MAX7219? https://github.com/rm-hull/luma.led_matrix
- Library for OLED: SH1106? https://github.com/rm-hull/luma.oled

Lippsson, 2024-05-16:
Some more ideas for display integration: 
[image]

Lippsson, 2024-05-16:
[image]

gschnitzler, 2024-06-21:
Have a look at [this implementation](https://github.com/gschnitzler/oled_phoniebox).
Its sort of a drop in replacement for splittis script. Just follow splittis install script, then install any missing python dependencies.
This might provide a better starting point for your own implementation.