MiczFlor/RPi-Jukebox-RFID #2375: BUG when cloning master

mzaiss, 2024-05-18:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/3431ba1a1b8865796fa618b8a0c4a257c842c25f/scripts/installscripts/install-jukebox.sh#L913

I suggest adding --depth 1 in the clone line.

I get this error otherwise:
[image]

mzaiss, 2024-05-18:
also pip module was not found for master installation.
i ran: 

sudo apt install python3-pip

to resolve

s-martin, 2024-05-18:
I think your connection broke while cloning, so it was not completed. Maybe try to put your box closer to the WiFi router.

I think all other errors are because of the not succeed cloning.

mzaiss, 2024-05-18:
Probably,
but I got this error repeatedly.
And --depth 1 solves it.
On the device there is also no reason not to not run  --depth 1, is it?

s-martin, 2024-05-18:
I agree depth 1 could be a good idea.

s-martin, 2024-05-19:
Maybe `--single-branch` in addition to the current command is useful, because it doesn’t fetch all the other branches.

s-martin, 2024-05-19:
This looks like a good overview: https://github.blog/2020-12-21-get-up-to-speed-with-partial-clone-and-shallow-clone/

AlvinSchiller, 2024-05-19:
I compared the commands for used space after initial clone with checked out "develop":
Full: ~44MB
NoBlob: ~29MB
NoTree: ~27MB
Depth 1: ~26MB

IF we want to make a change here, i also would vote for the `--filter=blob:none` option, as it has reduced size and the least side effects, with development usecases.

@mzaiss 
Nevertheless would be interesting, why this causes trouble in your setup, if it is not network related.

s-martin, 2024-06-04:
@mzaiss did you check again? I still suspect a temporary issue like bad WiFi or a server, which is down. But if the issue still persists for you, we should investigate deeper.

mzaiss, 2024-06-05:
i havent reinstalled since then, but suggest depth 1 as standard to save
time and memory.

s-martin, 2024-12-20:
We still assume the problem was related to a temporary network issue.

At this point we do not intend to change the current clone behavior.

Closing this issue now, feel free to reopen, if the problem comes up again.

closed 2024-12-20 by s-martin