MiczFlor/RPi-Jukebox-RFID #2387: 🐛 | RC522 irq not working properly

damaev, 2024-06-20:
### Version

3.5.3

### Branch

"future3/main", "v2/main"

### OS

Bullseye lite 32bit

### Pi model

Zero 2w

### Hardware

Mfrc522 

### What happened?

The Reader is reading just from time to time, so it sometimes takes a  looong time to read a card. Sometimes its fast.
I started having this issue after upgrading to a zero2 with some v3 Version half a year ago. Now i found the time to dig into it. Restarting with a plan v2 Version. Still haveing the issue.

What I tied:
Checking the soldering -> unchanged 
Stopping almost all jobs ->  unchanged 
Starting daemon by hand -> rst pin already in use -> change to unused -> issue persists 
Using my previous Version of phoniebox v2 from. 2 years ago -> issue persists
trying v3 with bullseye instead of bookworm -> issue persists

During that time I got the feeling the irq is not working. 
Changing irq pin -> issue persists
Reading irq pin by hand -> interrupts appear randomly -> changing mfrc522 board -> interrupts appear randomly -> issue persists 
Changing cables - issue persists 

I got the feeling that for some reason, the irq is not working (maybe anymore) as planned by the designer of the underlying rc522.py library 
Using another library without irq -> uid is read at hand

Digging into the Reader component of phoniebox again, I commented out the line "self.waiting_for_tag()"
Now I directly get the uid. -> **ISSUE SOLVED**

However, y am I the only one with this issue?
Will it  cause more trouble? 
Should an option "skip_irq" for rc522 readers be integrated. 

Maybe someone has the time and hardware to dig into it.

s-martin, 2024-06-20:
Can you post the log here?

damaev, 2024-06-20:
the error.log is empty. the app.log1 showing the delay between reading a card beeing in place all the time:
```
20.06.2024 16:06:03 -   19:readerbase.py      - jb.rfid.522(read_00) - read_00Thread   - INFO     - Initializing reader 'MFRC522 via SPI' from '/home/pi/RPi-Jukebox-RFID/src/jukebox/components/rfid/hardware/rc522_spi/rc522_spi.py'
20.06.2024 16:06:03 -   20:readerbase.py      - jb.rfid.522(read_00) - read_00Thread   - DEBUG    - Reader object is <components.rfid.hardware.rc522_spi.rc522_spi.ReaderClass object at 0x61b31bf8> for reader config key 'read_00'
20.06.2024 16:06:03 -   78:rc522_spi.py       - jb.rfid.522(read_00) - read_00Thread   - INFO     - Using legacy_mode = 'False'
20.06.2024 16:06:06 -  695:plugs.py           - jb.plugin.call       - StartJingle     - DEBUG    - Calling: volume.ctrl.set_volume(args=[40], kwargs=None)
20.06.2024 16:06:12 -  175:__init__.py        - jb.rfid(read_00)     - read_00Thread   - INFO     - Received card id = '1149614216389153'
20.06.2024 16:06:14 -  224:__init__.py        - jb.rfid(read_00)     - read_00Thread   - INFO     - Unknown card: '1149614216389153'
20.06.2024 16:07:02 -  111:server.py          - jb.rpc.server        - MainThread      - DEBUG    - Request: {'id': 'b4c4cc0c-96e1-46b0-9dc6-4bcd36e66ce1', 'package': 'cards', 'plugin': 'list_cards', 'method': None, 'kwargs': {}}
20.06.2024 16:07:02 -  695:plugs.py           - jb.plugin.call       - MainThread      - DEBUG    - Calling: cards.list_cards(args=(), kwargs={})
20.06.2024 16:07:02 -  151:server.py          - jb.rpc.server        - MainThread      - DEBUG    - Sending response: {'result': {}, 'id': 'b4c4cc0c-96e1-46b0-9dc6-4bcd36e66ce1'}
20.06.2024 16:07:48 -  175:__init__.py        - jb.rfid(read_00)     - read_00Thread   - INFO     - Received card id = '1149614216389153'
20.06.2024 16:07:48 -  224:__init__.py        - jb.rfid(read_00)     - read_00Thread   - INFO     - Unknown card: '1149614216389153'
20.06.2024 16:08:06 -  175:__init__.py        - jb.rfid(read_00)     - read_00Thread   - INFO     - Received card id = '1149614216389153'
20.06.2024 16:08:06 -  224:__init__.py        - jb.rfid(read_00)     - read_00Thread   - INFO     - Unknown card: '1149614216389153'
20.06.2024 16:08:42 -  175:__init__.py        - jb.rfid(read_00)     - read_00Thread   - INFO     - Received card id = '1149614216389153'
20.06.2024 16:08:42 -  224:__init__.py        - jb.rfid(read_00)     - read_00Thread   - INFO     - Unknown card: '1149614216389153'
20.06.2024 16:08:55 -  175:__init__.py        - jb.rfid(read_00)     - read_00Thread   - INFO     - Received card id = '1149614216389153'
20.06.2024 16:08:55 -  224:__init__.py        - jb.rfid(read_00)     - read_00Thread   - INFO     - Unknown card: '1149614216389153'
20.06.2024 16:09:28 -  175:__init__.py        - jb.rfid(read_00)     - read_00Thread   - INFO     - Received card id = '1149614216389153'
20.06.2024 16:09:28 -  224:__init__.py        - jb.rfid(read_00)     - read_00Thread   - INFO     - Unknown card: '1149614216389153'
20.06.2024 16:10:00 -  175:__init__.py        - jb.rfid(read_00)     - read_00Thread   - INFO     - Received card id = '1149614216389153'
20.06.2024 16:10:00 -  224:__init__.py        - jb.rfid(read_00)     - read_00Thread   - INFO     - Unknown card: '1149614216389153'
20.06.2024 16:10:13 -  175:__init__.py        - jb.rfid(read_00)     - read_00Thread   - INFO     - Received card id = '1149614216389153'
20.06.2024 16:10:13 -  224:__init__.py        - jb.rfid(read_00)     - read_00Thread   - INFO     - Unknown card: '1149614216389153'
20.06.2024 16:10:44 -  175:__init__.py        - jb.rfid(read_00)     - read_00Thread   - INFO     - Received card id = '1149614216389153'
20.06.2024 16:10:44 -  224:__init__.py        - jb.rfid(read_00)     - read_00Thread   - INFO     - Unknown card: '1149614216389153'
20.06.2024 16:11:18 -  175:__init__.py        - jb.rfid(read_00)     - read_00Thread   - INFO     - Received card id = '1149614216389153'
20.06.2024 16:11:18 -  224:__init__.py        - jb.rfid(read_00)     - read_00Thread   - INFO     - Unknown card: '1149614216389153'
20.06.2024 16:11:37 -  175:__init__.py        - jb.rfid(read_00)     - read_00Thread   - INFO     - Received card id = '1149614216389153'
20.06.2024 16:11:37 -  224:__init__.py        - jb.rfid(read_00)     - read_00Thread   - INFO     - Unknown card: '1149614216389153'
20.06.2024 16:11:54 -  175:__init__.py        - jb.rfid(read_00)     - read_00Thread   - INFO     - Received card id = '1149614216389153'
20.06.2024 16:11:54 -  224:__init__.py        - jb.rfid(read_00)     - read_00Thread   - INFO     - Unknown card: '1149614216389153'
20.06.2024 16:12:13 -  175:__init__.py        - jb.rfid(read_00)     - read_00Thread   - INFO     - Received card id = '1149614216389153'
20.06.2024 16:12:13 -  224:__init__.py        - jb.rfid(read_00)     - read_00Thread   - INFO     - Unknown card: '1149614216389153'
20.06.2024 16:13:26 -  175:__init__.py        - jb.rfid(read_00)     - read_00Thread   - INFO     - Received card id = '1149614216389153'
20.06.2024 16:13:26 -  224:__init__.py        - jb.rfid(read_00)     - read_00Thread   - INFO     - Unknown card: '1149614216389153'
20.06.2024 16:13:47 -  175:__init__.py        - jb.rfid(read_00)     - read_00Thread   - INFO     - Received card id = '1149614216389153'
20.06.2024 16:13:47 -  224:__init__.py        - jb.rfid(read_00)     - read_00Thread   - INFO     - Unknown card: '1149614216389153'
[... cut ...] ': None, 'kwargs': {}}
20.06.2024 21:06:06 -  695:plugs.py           - jb.plugin.call       - MainThread      - DEBUG    - Calling: host.get_autohotspot_status(args=(), kwargs={})
20.06.2024 21:06:06 -  151:server.py          - jb.rpc.server        - MainThread      - DEBUG    - Sending response: {'result': 'not-installed', 'id': '615a2d16-50d0-4d1e-92ca-f423511f5085'}
20.06.2024 21:34:43 -  263:server.py          - jb.pub.server        - PubServer       - INFO     - Un-subscription received
```

s-martin, 2024-06-21:
Just to understand your problem: 
It is working, but sometimes you have a huge delay between reading the card and the action triggered, correct?

could you also post your `rfid.yaml` and `cards.yaml`?

As I don’t have a rc522 reader, others may have more experience.

damaev, 2024-06-21:
I use a fresh system
Hence, Cards.yaml should be more or less plain.
I edited the rfid.yaml after opening this issue to allow placing instead of swiping.
Other than that it's like the setup-script configured it. So the issue used the autoconfig.
As I already mentioned, I also tested a plain v2. Even more, I did almost all testing on the v2/main setup. Like disabling all phoniebox services and starting `python daemon-rfid.py`which clearly showed the unsteady delays. 
Therefore, I conclude, it's either is a problem related to the pi zero2; To my soldering, which I doublechecked (ut who knows); There was an update to the rc522 library or mfrc522-partnever really worked correctly (from my short digging into the documentation of the rc522, the purpose of the irq may not necessarily indicate a nearby card)

damaev, 2024-06-21:
To clarify:
The reading of an uid seems to work pretty fine, like in no time. However the irq-signal is not dropping. Hence, the script keeps waiting for a tag even though a tag is present

flesser, 2024-06-22:
Can confirm, I have the same issue with a fresh install from master on a Raspberry 3B+ with RFID-RC522 connected to the default pins.

It hangs at `self.device.wait_for_tag()` ( https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/scripts/Reader.py.experimental#L91 which [gets copied to Reader.py be the setup script](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/components/rfid-reader/RC522/setup_rc522.sh#L32)).

If I comment out that line, tag IDs get read.

damaev, 2024-06-22:
Did it work correctly beforhand on the 3b+?
I was Using a pi zero 1. So it was always Slowenien, but I wouldn't say *that slow*.

9month ago the rc522.py library was updated to include a timeout during `wait_for_tag()` especially for our project.
https://github.com/ondryaso/pi-rc522/pull/70
Maybe this update is causing trouble now.

damaev, 2024-06-22:
i addad a fix with an additional config param `skip_irq`, see above

flesser, 2024-06-23:
Unfortunately with `skip_irq` (or commented out line) the "remove card to stop playing" does not work because `reader.readCard()` in https://github.com/damaev/RPi-Jukebox-RFID/blob/develop/scripts/daemon_rfid_reader.py#L88 now immediately returns even if there is no card, resulting in the alarm to be reset any read attempt, even if the cardid was None

s-martin, 2024-06-24:
From my point of view the original problem is tracked in #2373
I suggest to close this issue.

s-martin, 2024-11-07:
Closing. Follow up in #2373

closed 2024-11-07 by s-martin