MiczFlor/RPi-Jukebox-RFID #2388: Grouping media

heili77, 2024-06-23:
### Feature

I would like to group media like radio plays to reduce the number of cards.

### User perspective

The user can add a card associated with the group. Elements of the group could be chosen randomly or in order. Each group element can contain multiple tracks and plays all tracks in order as usual. 
So one could group all radio plays of „paw patrol“ or „die drei ???“ together and needs only one card. 

### Further information

Maybe a group could also just make a big playlist, with cards for each element and one for the group, like the random card does for all media.

heili77, 2024-06-23:
Currently one can only make a big playlist, where the borders of each radio play are not available or make a card for each single radio play, with just a view minutes of playtime.

s-martin, 2024-06-23:
As @heili77 said you could create a big playlist at the moment.

I guess you checked this?
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/future3/develop/documentation/builders/webapp/playlists-livestreams-podcasts.md