MiczFlor/RPi-Jukebox-RFID #2402: 🐛 | bookworm: check for raspian not working

mittler-works, 2024-07-13:
### Version

3.5.3

### Branch

future3/main

### OS

Raspberry Pi OS Bookworm Lite

### Pi model

Zero 2W

### What happened?

When running the installer script, it fails saying

```
ERROR: '<something>' found more than once in 'unknown'
```

Where `<something>` is not always the same thing.

I tracked it down, the problem is the check for Raspian [here](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/0ab75c1d28955e85c2069503b5a8428a69967d0d/installation/includes/02_helpers.sh#L79).

Newer releases of Raspberry Pi OS (bookworm) do not display `raspian` in `/etc/os-release` no more, but just `debian`. So there is no way to really tell if we are on Raspberry Pi OS.
I have found articles that propose to check the existence of `/etc/apt/sources.list.d/raspi.list` to determine if we are on Raspberry Pi OS. Although this is obviously not perfect either, it might be a better solution than consulting `/etc/os-release` for the future.

mittler-works, 2024-07-14:
This only seems to affect bookworm 64bit. bookworm 32 bit shows raspian, as usual.

s-martin, 2024-07-14:
At this time 64bit is not supported (there's also a check for it).

mittler-works, 2024-07-15:
I know that, but I'm always interested in trying things out. ;-) 
Basically, bookworm/64bit is already running quite decent. It's just little things; but I think it's important to track them, even if the software is _currently_ not supporting it. That might save a bit of work for `future` later. :wink:

s-martin, 2024-07-15:
> Basically, bookworm/64bit is already running quite decent. It's just little things; but I think it's important to track them, even if the software is _currently_ not supporting it. That might save a bit of work for `future` later. :wink: 

I agree. 

We also happily discuss possible changes for a working 64bit implementation, so if anybody opens a PR with necessary changes interested people could review it.
I know @alesrosina did also make it work (see #2313)

alesrosina, 2024-07-15:
> > Basically, bookworm/64bit is already running quite decent. It's just little things; but I think it's important to track them, even if the software is _currently_ not supporting it. That might save a bit of work for `future` later. 😉
> 
> I agree.
> 
> We also happily discuss possible changes for a working 64bit implementation, so if anybody opens a PR with necessary changes interested people could review it. I know @alesrosina did also make it work (see #2313)

Yes, basically I just removed all checks in setup script. And then it's running as it should on 64bit.

I probably should take the time to make a PR, but probably not before autumn, since in summer time i prefer to be outside in my free time ;)

mittler-works, 2024-07-15:
I can prepare a PR to lift the 64 bit restriction. It would only be a change in the `install-jukebox.sh` (removing [_check_os_type](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/future3/main/installation/install-jukebox.sh#L89-L104)) and the thing with the Raspian check. There I would then use an or-condition in terms of "Either the os-release property ID contains raspian or the file /etc/apt/sources.list.d/raspi.list exists and contains corresponding raspian sources".

I've found no other issues with 64 bit. The GPIO Problems with kernel 6.6 upwards mentioned in the other issue are not specific to 64 bit, but also affect 32 bit, right? So that wouldn't be a show stopper for lifting 64 bit restriction, I think.

alesrosina, 2024-07-15:
> I can prepare a PR to lift the 64 bit restriction. It would only be a change in the `install-jukebox.sh` (removing [...]

Yes, from my experience that's correct :+1:

harrydoddnoble, 2024-09-15:
@mittler-works did you manage to create a PR for this?

mittler-works, 2024-09-16:
Sorry, that got lost in the daily grind. :wink: 
I have just created the PR. :wave:

harrydoddnoble, 2024-09-16:
> Sorry, that got lost in the daily grind. 😉
> I have just created the PR. 👋

Thank you!

mittler-works, 2024-09-17:
With pleasure :)

But one other thing I would like to discuss: the error message you get when you are not on Raspberry Pi OS is not meaningful at all (see OP of this issue).

The problem origins in [getting the boot and cmdline file paths](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/future3/develop/installation/includes/02_helpers.sh#L91-L116). When the raspbian check (maybe it should be Raspberry Pi OS Check today? Thats also the reason why there is not always "raspbian" in `/etc/os-release`) returns `false`, then `_get_boot_file_path` just returns `unknown` instead of stopping the execution with a failure. So the routine continues as usual and if you want to write or do something with the returned path, it fails with curious error messages.

My proposal would be, that the function should write a meaningful error message to stderr and exit with non-zero exit code. I don't know if there is any kind of logger and/or exception handling implemented in the installer scripts. If not, it would be as easy as echoing to stderr and exit.

s-martin, 2024-09-17:
> My proposal would be, that the function should write a meaningful error message to stderr and exit with non-zero exit code. I don't know if there is any kind of logger and/or exception handling implemented in the installer scripts. If not, it would be as easy as echoing to stderr and exit.

yes, we do.

please check here

https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/f2a1730a8df693350aef14aa6df6f563ffe6f786/installation/install-jukebox.sh#L42-L87

s-martin, 2024-11-07:
Fixed in #2425

PR #2425 (2024-11-07): Future3/lift 64bit restriction
Closes #2402 

This PR removes the check for 64bit OS in the installer script.

This PR extends the check for raspian OS to an alternate condition, which is the presence of `/etc/apt/sources.list.d/raspi.list` as some installations of raspian displaying `debian` instead of `raspian` in `/etc/os-release`.

Related:
#2402 
#2041 
#2313
closed 2024-11-07 by s-martin