MiczFlor/RPi-Jukebox-RFID #2405: 🐛 | RC522 cards not detected

oliboss, 2024-07-25:
### Version

3.5.3

### Branch

future3/main

### OS

Raspbian Bookworm 12 - 32 bit

### Pi model

Raspberry Pi 4 

### Hardware

* RFID reader RC533
* HiFi AMP HAT TAS5713 Amplifier Audio Module

### What happened?

**Current behavior:**
* Jukebox is working in terms of playback, library, etc. 
* After using `break-system-packages` on pip installs the RFID reader could be selected and the install script installed all python dependencies
* RFID reader is connected according to the schema on the website
* This simple python script shows the ids from different cards / tokens
```python
#!/usr/bin/env python3
import RPi.GPIO as GPIO
from mfrc522 import SimpleMFRC522
reader = SimpleMFRC522()

try:
        id, text = reader.read()
        print(id)
        print(text)
finally:
        GPIO.cleanup()
```
* When going in the web interface, select "add a card" and presenting a card nothing happens

**Expected behavior:**
* Card should be recognized

### Logs

error.log.1
```
25.07.2024 16:42:19 -  650:plugs.py           - jb.plugin            - ShutdownThread  - ERROR    - Ignoring failed package atexit function: 'rfid.atexit()'
25.07.2024 16:42:19 -  651:plugs.py           - jb.plugin            - ShutdownThread  - ERROR    - Reason: AttributeError: 'NoneType' object has no attribute 'stop'
25.07.2024 16:42:19 -  652:plugs.py           - jb.plugin            - ShutdownThread  - ERROR    - Detailed reason:
Traceback (most recent call last):
  File "/home/user/RPi-Jukebox-RFID/src/jukebox/jukebox/plugs.py", line 648, in close_down
    results.append(func(**kwargs))
                   ^^^^^^^^^^^^^^
  File "/home/user/RPi-Jukebox-RFID/src/jukebox/components/rfid/reader/__init__.py", line 264, in atexit
    reader.stop()
  File "/home/user/RPi-Jukebox-RFID/src/jukebox/components/rfid/reader/__init__.py", line 136, in stop
    self._reader.stop()
    ^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'stop'

```

### Configuration

rfid.yaml
```yaml
rfid:
  readers:
    read_00:
      module: rc522_spi
      config:
        spi_bus: 0
        spi_ce: 0
        pin_irq: 24
        pin_rst: 25
        mode_legacy: false
        antenna_gain: 4
        log_all_cards: false
      same_id_delay: 1.0
      log_ignored_cards: false
      place_not_swipe:
        enabled: false
        card_removal_action:
          alias: pause

```

plimptm, 2024-07-26:
Have you read through [this issue  thread](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/2313) for bookworm related issues? There are known issus with gpio.

oliboss, 2024-07-28:
Thanks a lot for the quick reply.

I got it to work with the lpio replacement but it stopped working soon after - which most likely is due to some other issue.

closed 2024-07-28 by oliboss