MiczFlor/RPi-Jukebox-RFID #2418: No audio output with Spotify

Guggi11, 2024-08-17:
### Version

v.2.7.0

### Branch

Master

### OS

Raspberry Pi OS Legacy 64-Bit Lite

### Pi model

4B

### Hardware

MiniAmp
RC522
1 speaker (Mono setup)

### What happened?

In the WebApp I can start Iris and see all my Playlists. When i want to play one, the URLs are loaded and I also see the title in the WebApp. But I can not play a title. Sound is working and local files, webradio etc. is working.
I found an error with the HTTP and MPD ports.

### Logs

ERROR    2024-08-17 02:38:15,320 [13260:MainThread] mopidy.commands
  Frontend (MpdFrontend) initialization error: MPD server startup failed: [Errno 98] Address already in use
INFO     2024-08-17 02:38:15,321 [13260:IrisFrontend-7] mopidy.internal.path
  Creating dir file:///root/.local/share/mopidy/iris
DEBUG    2024-08-17 02:38:15,322 [13260:MainThread] mopidy.http.actor
  Starting HTTP server
ERROR    2024-08-17 02:38:15,324 [13260:MainThread] mopidy.commands
  Frontend (HttpFrontend) initialization error: HTTP server startup failed: [Errno 98] Address already in use

Guggi11, 2024-08-19:
Is it possible that it is because of the 64Bit Lite Version?

dmuehlemann, 2024-08-19:
I'm experiencing the exact same issue. I did a fresh install today, but I'm unable to play anything on Spotify. I'm using the 32-bit Lite version.

Guggi11, 2024-08-19:
Now it started once via iris. One playlist (I dont know why) started playing. But now again it didt'n work

s-martin, 2024-08-19:
There seems to be again an issue with Mopidy-Spotify, see https://github.com/pimoroni/pirate-audio/issues/98#issuecomment-2292192728

Maybe this is related

s-martin, 2024-11-10:
Fixed in #2435 

Fixed in release [2.8.0](https://github.com/MiczFlor/RPi-Jukebox-RFID/releases/tag/v2.8.0)

closed 2024-11-10 by s-martin