MiczFlor/RPi-Jukebox-RFID #2424: 🐛 | Raspberry Pi 5 - 64 bit Kernel

harrydoddnoble, 2024-09-15:
### Version

3

### Branch

main

### OS

Raspberry Pi OS Bullseye lite - 32bit

### Pi model

5

### What happened?

Raspberry PI 5 cannot run a 32 bit kernel so even if you change the config file it will still run 64 bit. Is there any way to install Jukebox on a RaspberryPI 5?

s-martin, 2024-09-16:
Please check here https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/2041#issuecomment-1615587740 for a temporary workaround.

we will probably remove the restriction to 32bit soon, see also

harrydoddnoble, 2024-09-17:
This worked when removing the check. There were some issues with the version of ZeroMQ 4.3.5 as it wanted to install 4.3.4 so I installed 4.3.5 first and then pyzmq version < 26

closed 2024-09-17 by harrydoddnoble