MiczFlor/RPi-Jukebox-RFID #2426: 🐛 | PirateHAT Display Issues

jszust, 2024-09-16:
### Version

2.7.0

### Branch

master

### OS

Raspberry Pi OS Lite - 32 Bit Bullseye

### Pi model

Zero 1

### Hardware

Raspberry Pi Zero WH + Pirate Audio Mini-Amp with Display for Headphones + PiSugar3 + HiLetGo USB RFID Reader

### What happened?

I managed to get RPi-Jukebox installed (Spotify Edition) alongside the Pirate Audio mini-amp and have been having issues with the display. I tested the display with a mopidy-only install and confirmed that it, along with the Pirate HAT GPIO buttons, all worked well without issue. 

I used the following procedure to have RPi-Jukebox work with the Pirate HAT:

1) sudo apt update && upgrade on a fresh install
2) Installed RPi-Jukebox Spotify edition
3) Modified /boot/config.txt to:
include:
- dtoverlay=hifiberry-dac
- gpio=25=op,dh
- dtparam=i2c_arm=on
- dtparam=i2s=on
- dtparam=spi=on

Comment out:
- dtparam=audio=on

Modify:
- dtoverlay=vc4-kms-v3d**,noaudio**

4) Reboot
5) sudo apt-get install the following python dependencies:
- python3-rpi.gpio 
- python3-spidev 
- python3-pip 
- python3-pil 
- python3-numpy

6) Install the following numpy dependency for the ST7789 library: sudo apt-get install libopenblas-dev

7) Stop the GPIO Control Service with the following:
sudo systemctl stop phoniebox-gpio-control.service
sudo systemctl disable phoniebox-gpio-control.service

8) Modify /etc/mpd.conf to the following:
```
# An example of an ALSA output:
audio_output {
        enabled         "yes"
        type            "alsa"
        name            "HiFiBerry DAC+ Lite"
        device          "hifiberry"
        auto_resample   "no"
        auto_channels   "no"
        auto_format     "no"
        dop             "no"
}
#
audio_output {
        type            "alsa"
        name            "My ALSA Device"
#       device          "hw:0,0"        # optional
#       mixer_type      "hardware"      # optional
#       mixer_device    "default"       # optional
        mixer_control   "Master"                # optional
#       mixer_index     "0"             # optional
```

9) Make sure SPI is activated even though the config boot line should enable it:
sudo raspi-config nonint do_spi 0

10) Ensure iris is given root privileges consistent with Pirate Hat installation:
echo "mopidy ALL=NOPASSWD: /usr/local/lib/python3.9/dist-packages/mopidy_iris/system.sh" | sudo tee -a /etc/sudoers

11) Install Pirate Display plugins:
sudo pip3 install Mopidy-PiDi pidi-display-pil pidi-display-st7789 mopidy-raspberry-gpio

12) modify /etc/mopidy.conf to map the GPIO buttons and enable the display:
```
[raspberry-gpio]
enabled = true
bcm5 = play_pause,active_low,150
bcm6 = volume_down,active_low,150
bcm16 = next,active_low,150
bcm20 = 
bcm24 = volume_up,active_low,150

[pidi]
enabled = true
display = st7789
```

13) Reboot

It took a lot of trial and error to get these two devices to play nice, but following these steps I was able to get it working 100%, except that now the display is VERY temperamental. If I quickly volume up or down, the display goes blank. If you skip through songs too quickly using the RFID cards, it will become discolored and eventually go blank. The song progress bar does not update automatically. Sometimes the display doesn't start up at all. I can confirm the display worked well with a mopidy-only install, but mopidy does not have a nice RFID plugin akin to this project. I'm building this jukebox for my children so I want the GPIO buttons to be usable and for the display to work. I'm wondering if anyone else has dealt with this and has any ideas? Using systemctl restart mopidy will reset the display and bring it back to life, but the same issues persist.

s-martin, 2024-09-16:
#1112 related

s-martin, 2024-09-18:
Thanks for this howto!

jszust, 2024-11-04:
@s-martin just wanted to follow up. I upgraded our phonieboxes from Pi Zero W to the Pi Zero 2 W and this is working **flawlessly** now. The screen is super responsive, tracks change instantaneously, and it is a complete treat. My kids absolutely love it and I cannot thank you and all the contributors enough for putting this all together. I updated my initial post with some additional steps to get it 100% there. Thanks!