MiczFlor/RPi-Jukebox-RFID #2429: 🐛 | Can't play content: GStreamer error: Could not get/set settings from/on resource

The-Judge, 2024-09-18:
### Version

2.7.0 - 550a2583 - master

### Branch

master

### OS

Raspbian GNU/Linux 11 (bullseye) - Lite 32Bit

### Pi model

3 B Rev 1.2

### Hardware

* Raspberry Pi 3B
* HifiBerry DAC+

### What happened?

Installing Phoniebox works fine. When the box boots up, I can hear the boot-sound from the speakers fine also.
But when I play content via the WebUI or Iris, nothing plays and the following is written to the mopidy log:

ERROR    [MainThread] mopidy.audio.gst GStreamer error: Could not get/set settings from/on resource.

### Logs

ERROR    [MainThread] mopidy.audio.gst GStreamer error: Could not get/set settings from/on resource.

### Configuration

# mpd.conf

```
music_directory		"/home/pi/RPi-Jukebox-RFID/shared/audiofolders"
playlist_directory		"/home/pi/RPi-Jukebox-RFID/playlists"
db_file			"/var/lib/mpd/tag_cache"
log_file			"syslog"
pid_file			"/run/mpd/pid"
state_file			"/var/lib/mpd/state"
sticker_file                   "/var/lib/mpd/sticker.sql"
user				"root"
bind_to_address		"localhost"
log_level			"default"
auto_update    "yes"
auto_update_depth "10"
input {
        plugin "curl"
}
input {
        enabled    "no"
        plugin     "qobuz"
}
input {
        enabled      "no"
        plugin       "tidal"
}
decoder {
        plugin                  "hybrid_dsd"
        enabled                 "no"
}
audio_output {
	type		"alsa"
	name		"My ALSA Device"
	mixer_control	"DSP Program"		# optional
}
volume_normalization		"yes"
filesystem_charset		"UTF-8"
```

# /etc/mopidy/mopidy.conf

```
[core]
cache_dir = /var/cache/mopidy
config_dir = /etc/mopidy
data_dir = /var/lib/mopidy
max_tracklist_length = 10000
restore_state = false

[logging]
verbosity = 0
format = %(levelname)-8s [%(threadName)s] %(name)s %(message)s
color = false
config_file =

[audio]
mixer = software
mixer_volume = 30
output = alsasink
buffer_time =

[proxy]
scheme =
hostname =
port =
username =
password =

[file]
enabled = true
media_dirs =
  $XDG_MUSIC_DIR|Music
  ~/|Home
excluded_file_extensions =
  .directory
  .html
  .jpeg
  .jpg
  .log
  .nfo
  .pdf
  .png
  .txt
  .zip
show_dotfiles = false
follow_symlinks = false
metadata_timeout = 1

[http]
enabled = true
hostname = 0.0.0.0
port = 6680
zeroconf = Mopidy HTTP server on $hostname
allowed_origins =
csrf_protection = true
default_app = mopidy

[m3u]
enabled = true
base_dir =
default_encoding = UTF-8
default_extension = .m3u
playlists_dir = /home/pi/RPi-Jukebox-RFID/playlists

[softwaremixer]
enabled = true

[stream]
enabled = true
protocols =
  http
  https
  mms
  rtmp
  rtmps
  rtsp
metadata_blacklist =
timeout = 5000

[iris]
enabled = true
country = DE
locale = de_DE
verify_certificates = true
snapcast_enabled = true
snapcast_host = localhost
snapcast_port = 1780
snapcast_ssl = false
snapcast_stream = Default
spotify_authorization_url = https://jamesbarnsley.co.nz/iris/auth_spotify.php
lastfm_authorization_url = https://jamesbarnsley.co.nz/iris/auth_lastfm.php
genius_authorization_url = https://jamesbarnsley.co.nz/iris/auth_genius.php
data_dir = $XDG_DATA_DIR/iris

[local]
enabled = true
max_search_results = 100
media_dir = /home/pi/RPi-Jukebox-RFID/shared/audiofolders
scan_timeout = 1000
scan_flush_threshold = 100
scan_follow_symlinks = false
included_file_extensions =
excluded_file_extensions =
  .conf
  .jpg
  .txt
  placeholder
directories =
  Albums                  local:directory?type=album
  Artists                 local:directory?type=artist
  Composers               local:directory?type=artist&role=composer
  Genres                  local:directory?type=genre
  Performers              local:directory?type=artist&role=performer
  Release Years           local:directory?type=date&format=%25Y
  Tracks                  local:directory?type=track
  Last Week's Updates     local:directory?max-age=604800
  Last Month's Updates    local:directory?max-age=2592000
timeout = 10
use_artist_sortname = false
album_art_files =
  *.jpg
  *.jpeg
  *.png

[mpd]
enabled = true
hostname = 0.0.0.0
port = 6600
password =
max_connections = 20
connection_timeout = 60
zeroconf = Mopidy MPD server on $hostname
command_blacklist =
  listall
  listallinfo
default_playlist_scheme = m3u

[spotify]
enabled = true
username = <REMOVED>
password = ********
client_id = c9c9350b-<REMOVED>
client_secret = ********
bitrate = 160
volume_normalization = true
timeout = 10
allow_cache = true
cache_size = 8192
allow_playlists = true
search_album_count = 20
search_artist_count = 10
search_track_count = 50
```

harrydoddnoble, 2024-09-18:
I ma having the same issue - not sure if it's related to [this](https://github.com/mopidy/mopidy-spotify/issues/394)?

s-martin, 2024-09-18:
Yes, the problem is probably that Spotify requires now oauth authentication.

This has to be fixed in [mopidy-spotify](https://github.com/mopidy/mopidy-spotify)

s-martin, 2024-10-07:
**All**

PR #2435 updates mopidy-spotify, which should fix this issue.

Everyone is invited to test this fix.

seschu, 2024-10-13:
Hi there I have the same problem I think. i am confused a little bit: How do I get the credentials.json mentioned in the fix?

s-martin, 2024-10-13:
@seschu 

The PR #2345 is supposed to fix this, please check it out

seschu, 2024-10-13:
> @seschu
> 
> The PR #2435 is supposed to fix this, please check it out

I did update the spotify mopidy version but now I am getting this error

"
Oct 13 20:57:52 theobox mopidy[6044]: INFO     [MpdSession-38] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:47336
Oct 13 20:57:52 theobox mopidy[6044]: INFO     [MpdSession-39] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:47340
Oct 13 20:57:52 theobox mopidy[6044]: INFO     [MpdSession-40] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:47346
Oct 13 20:57:52 theobox mopidy[6044]: Traceback (most recent call last):
Oct 13 20:57:52 theobox mopidy[6044]:   File "/usr/local/lib/python3.9/dist-packages/mopidy/audio/actor.py", line 577, in _on_source_setup
Oct 13 20:57:52 theobox mopidy[6044]:     self._source_setup_callback(source)
Oct 13 20:57:52 theobox mopidy[6044]:   File "/usr/local/lib/python3.9/dist-packages/mopidy_spotify/backend.py", line 50, in on_source_setup
Oct 13 20:57:52 theobox mopidy[6044]:     source.set_property("access-token", self.backend._web_client.token())
Oct 13 20:57:52 theobox mopidy[6044]: TypeError: object of type `GstSpotifyAudioSrc' does not have property `access-token'
Oct 13 20:57:52 theobox mopidy[6044]: ERROR    [MainThread] mopidy.audio.gst GStreamer error: Could not get/set settings from/on resource.

"

s-martin, 2024-10-13:
Probably not completely fixed

https://github.com/mopidy/mopidy-spotify/issues/394#issuecomment-2401706956

seschu, 2024-10-13:
I think we have to get the credentials.json file from librespot https://github.com/mopidy/mopidy-spotify/issues/394

seschu, 2024-10-13:
I made it work by doing the following
generateing a new request token here https://developer.spotify.com/documentation/web-playback-sdk/tutorials/getting-started
then updating the modiby client id and secret (dont know if this is necessary
I then created a credentials.json file in this location /var/lib/mopidy/spotify/credentials-cache/credentials.json
The file has this format 
{"username":"weirdusername",
        "auth_type":1,"auth_data":"generatedaccesstoken"}
generatedaccesstoken is the token received from the spotify website
the username is a weird random combination of letter sthat I see when I check my spotify profile page and press edit profile.

I then updated the gstreamerplugin with this alpha release https://github.com/kingosticks/gst-plugins-rs-build/releases/tag/gst-plugin-spotify_0.14.0-alpha.1-1
Here I only downlaoded this file https://github.com/kingosticks/gst-plugins-rs-build/releases/download/gst-plugin-spotify_0.14.0-alpha.1-1/gst-plugin-spotify_0.14.0.alpha.1-1_armhf.deb
and I installed it doing
sudo apt remove gst-plugin-spotify
sudo dpkg -i gst-plugin-spotify_0.14.0.alpha.1-1_armhf.deb

I am not sure whether all steps were necessary. It works currently.

jonaskor, 2024-10-28:
> I think we have to get the credentials.json file from librespot [mopidy/mopidy-spotify#394](https://github.com/mopidy/mopidy-spotify/issues/394)

There seems to be some progress in the mentioned thread, any chance to get a fix for this here also?

kiwitea, 2024-10-29:
> I made it work by doing the following:
> 
>     1. I had to generate a new request token directly at Spotify
>        https://developer.spotify.com/documentation/web-playback-sdk/tutorials/getting-started
> 
>     2. I updated the mopidy client id and secret (I actually don't know if this is necessary) under /etc/mopidy/mopidy.conf
> 
>     3. The new request token from step 1 is used for creating a file called credentials.json. I placed this file with sudo cp in this location /var/lib/mopidy/spotify/credentials-cache/credentials.json
>        The file has this content
>        {"username":"weirdusername",
>        "auth_type":1,"auth_data":"generatedaccesstoken"}
>        generatedaccesstoken is the token received from the spotify website in step 1.
>        the username is a weird random combination of letters that I see when I check my spotify profile page (on desktop in browser or in a desktop app) and press edit profile.
> 
>     4. I then updated the gstreamerplugin with this alpha release https://github.com/kingosticks/gst-plugins-rs-build/releases/tag/gst-plugin-spotify_0.14.0-alpha.1-1
>        Here I only downlaoded this file https://github.com/kingosticks/gst-plugins-rs-build/releases/download/gst-plugin-spotify_0.14.0-alpha.1-1/gst-plugin-spotify_0.14.0.alpha.1-1_armhf.deb
>        and I installed it doing
>        sudo apt remove gst-plugin-spotify
>        sudo dpkg -i gst-plugin-spotify_0.14.0.alpha.1-1_armhf.deb
> 
> 
> I am not sure whether all steps were necessary. It works currently.

Did the same and it works for me as well.

In addition to the above, I also had to install the `5.0.0a3` version of the [mopidy-spotify plugin](https://github.com/mopidy/mopidy-spotify/releases/tag/v5.0.0a3) which I didn't have yet. This enables the new authentication method and now you can remove the username/password from mopidy.conf.
```bash
sudo python3 -m pip install Mopidy-Spotify==5.0.0a3
```

Btw, I found it useful to check the logs from mopidy, to get quick feedback on whether authentication works or not and why
```bash
journalctl -u mopidy -b --grep spotify.web
```

PR #2435 (2024-11-04): Fix Spotify authentication 
This should fix #2429.

- [x] gst plug-in needs to be updated in install-jukebox.sh (gst-plugin-spotify_0.14.0.alpha.1-1_armhf.deb), see also https://github.com/mopidy/mopidy-spotify/discussions/405#discussioncomment-10881114

Everyone experiencing this issue is invited to test this fix.
closed 2024-11-04 by s-martin