MiczFlor/RPi-Jukebox-RFID #2436: 🐛 | ERROR: pyzmq is not installed

Sankara49, 2024-10-07:
### Version

?

### Branch

future3/main

### OS

Raspberry Pi OS Lite (Legacy, 32-bit)

### Pi model

Raspberry Pi 3 Model B Rev 1.2

### Hardware

nothing right now.
plan on using neuftech RFID Reader.
plan not to use GPIO in first iteration.

### What happened?

following the steps from Page "Installing Phoniebox future3"
Install Raspberry Pi OS Lite - **success** 
Pre-boot preparation - not done
Pre-install preparation / workarounds - not done
Install Phoniebox software - unsuccess
using the given one line command for stable release

I tried this procedere 2 times and it stops right at the same point. (with different download progress for Cython XX/10.7 MB)

### Logs

******************SHORT*****************

Set default raspi-config
Updating Raspberry Pi OS
Install Git & init repository
Install Jukebox Core
  Install Jukebox OS dependencies
  Install Python requirements
  Install pyzmq with libzmq-drafts to support WebSockets
  Copy PulseAudio configuration
  Register Jukebox settings
  Register Jukebox Core user service

****************************************
ERROR OCCURRED!
A non-recoverable error occurred.
Check install log for details:
/home/pi/INSTALL-1728334506.log
****************************************
ERROR: pyzmq is not installed
****************************************

******************LONG******************
(...)
  Collecting cython>=0.29
    Downloading https://www.piwheels.org/simple/cython/Cython-3.0.11-cp39-cp39-linux_armv7l.whl (10.7 MB)
       ━━━━━━━━━━━                              3.0/10.7 MB 62.4 kB/s eta 0:02:04
  ERROR: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, 
please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.
      cython>=0.29 from https://www.piwheels.org/simple/cython/Cython-3.0.11-cp39-cp39-linux_armv7l.whl#sha256=28d7a391c14b703ec140fdb8101cfc2b3134f21c0d7bbc553ce49965c6f95d6c:
          Expected sha256 28d7a391c14b703ec140fdb8101cfc2b3134f21c0d7bbc553ce49965c6f95d6c
               Got        bde8470ced1691d368e25eacf7376c2e86aaf78a249a17e3b017cf9bbbb9180f

  error: subprocess-exited-with-error
  
  × pip subprocess to install build dependencies did not run successfully.
  │ exit code: 1
  ╰─> See above for output.
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  full command: /home/pi/RPi-Jukebox-RFID/.venv/bin/python3 /home/pi/RPi-Jukebox-RFID/.venv/lib/python3.9/site-packages/pip/__pip-runner__.py install --ignore-installed --no-user --prefix /tmp/pip-build-env-zrrju6ht/overlay --no-warn-script-location --disable-pip-version-check -v --no-binary pyzmq --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple -- setuptools 'setuptools_scm[toml]' wheel packaging 'cffi; implementation_name == '"'"'pypy'"'"'' 'cython>=0.29; implementation_name == '"'"'cpython'"'"'' 'cython>=0.29.35; implementation_name == '"'"'cpython'"'"' and python_version >= '"'"'3.12'"'"''
  cwd: [inherit]
  Installing build dependencies: finished with status 'error'
error: subprocess-exited-with-error

× pip subprocess to install build dependencies did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
  Copy PulseAudio configuration
  Register Jukebox settings
  Register Jukebox Core user service
Created symlink /home/pi/.config/systemd/user/default.target.wants/jukebox-daemon.service -> /usr/lib/systemd/user/jukebox-daemon.service.

  -------------------------------------------------------
  Check installation

  Verify packages are installed: 'at alsa-utils caps espeak ffmpeg libasound2-dev mpg123 pulseaudio pulseaudio-module-bluetooth pulseaudio-utils python3 python3-venv python3-dev rsync'
  CHECK
  Verify '/home/pi/RPi-Jukebox-RFID/.venv' exists
  CHECK
  Verify modules are installed: 'pyzmq setuptools wheel evdev pyalsaaudio pulsectl python-mpd2 ruamel.yaml python-slugify requests tornado RPi.GPIO gpiozero flake8 pytest pytest-cov mock pydoc-markdown'

****************************************
ERROR OCCURRED!
A non-recoverable error occurred.
Check install log for details:
/home/pi/INSTALL-1728334506.log
****************************************
ERROR: pyzmq is not installed
****************************************
Abort!

### More info

Hello, this is my first interaction on Github and I hope I've done it right.
I was trying to install future3/main over the last 3-4 days and its ends up with an error everytime. One time - I can remember - it was failing to download the WebApp, but the error I can reproduce is the abouve one with not matching Hash at Cython.
I am looking forward for any advise.
Thanks a lot!
Sankara

Sankara49, 2024-10-11:
Problem Solved.
seems to be a problem with the source, whats fixed now on their side.

loerchd89, 2024-12-08:
I am facing the excact same issue. Any ideas how to solve?

s-martin, 2024-12-08:
@loerchd89 can you post the log file of the installation?

It's located in the Phoniebox directory and starts with `INSTALL` and a timestamp with the ending `.log`.

Sunc4stle, 2024-12-11:
I also have this problem.
OS: raspios-bullseye-armhf-lite
log: [INSTALL-1733954955.log](https://github.com/user-attachments/files/18103164/INSTALL-1733954955.log)

pabera, 2024-12-17:
@Sunc4stle Which Pi do you use?
Can you try with Bookworm Lite 64bit?

loerchd89, 2024-12-26:
@s-martin @pabera Sorry for the late response, I was busy with christmas :). I have tried now with Bookworm Lite 64bit - this did the trick! Thank you and merry Xmas

closed 2024-12-27 by pabera

AlvinSchiller, 2024-12-28:
I tried the installation a few days ago on a Pi 3A+ with Raspberry Pi OS (Legacy) Lite (raspios-bullseye-armhf-lite) and its working again.