MiczFlor/RPi-Jukebox-RFID #2438: 🐛 |

thesilversverker, 2024-10-10:
### Version

3.5.3

### Branch

future3/main

### OS

Raspbian GNU/Linux 12 (bookworm)

### Pi model

3B+

### Hardware

[ADAFruit PN532 RFID/NFC Shield](https://www.adafruit.com/product/789)

### What happened?

Installed via one-liner 

### Logs

[INSTALL-1728397540.log](https://github.com/user-attachments/files/17331484/INSTALL-1728397540.log)
[app.log](https://github.com/user-attachments/files/17331485/app.log)

### Configuration

[jukebox.txt](https://github.com/user-attachments/files/17331509/jukebox.txt)
[rfid.txt](https://github.com/user-attachments/files/17331510/rfid.txt)

### More info

Performing install & test with NFC tools seemed to work, did this to validate testing of soldered connections & wiring. Required *manual* disable/enable of I2C for detection.

[nfc_test.txt](https://github.com/user-attachments/files/17331526/nfc_test.txt)

thesilversverker, 2024-10-10:
I tried testing by scanning card, without seeing any activity in the app.log. I also used the webUI to navigate to cards, add card, and then scanned the card on the PN532 without any reaction I could see.

Fiete123, 2024-10-14:
Take a look to the feature list [527716](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/future3/main/documentation/developers/status.md#cards) the function via WEBUI is not finished yet

closed 2024-10-15 by thesilversverker