MiczFlor/RPi-Jukebox-RFID #2444: RDM6300 configuration bug

Darkcast, 2024-10-21:
### Version

???

### Branch

future/main

### OS

Raspberry pi os 32 lite 

### Pi model

Raspberry pi 4 

### Hardware

RDM6300

### What happened?

when I try to install using the following command ```cd; bash <(wget -qO- https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/future3/main/installation/install-jukebox.sh)``` when i get to the option to choose auto configure it just get stuck in this screen. ive done this like a thounsand times now :(

[image]

any help would be greatly apreciated.

s-martin, 2024-10-22:
Looks like there's again a behavior change with bookworm.

We need to change the disabling of the serial access, see https://github.com/raspberrypi/bookworm-feedback/issues/114#issuecomment-1776113246

Needs to be changed here https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/f2a1730a8df693350aef14aa6df6f563ffe6f786/src/jukebox/components/rfid/hardware/rdm6300_serial/setup.inc.sh#L6

We probably need to check, if we need to keep the current command for older OS versions or if this is backwards compatible.

Darkcast, 2024-10-22:
Thank you for looking into this issue :)

s-martin, 2024-11-07:
Has been fixed in #2450

PR #2450 (2024-11-07): Use correct raspi-config command for bookworm
- Fixes #2444 - original problem see https://github.com/raspberrypi/bookworm-feedback/issues/114#issuecomment-1776113246
- add function to check bookworm
closed 2025-05-27 by s-martin