MiczFlor/RPi-Jukebox-RFID #2445: 🚀 | Support request for PN532 V2.0 USB variant

Darkcast, 2024-10-23:
### Feature

Could you please add support for the PN532 V2.0 USB variant 

https://banggood.onelink.me/zMT7/ayw6bb2e

[image]

Thank you kindly 👻

s-martin, 2024-10-24:
There is an option to use USB, see https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/future3%2Fmain/documentation%2Fdevelopers%2Frfid%2Fgenericusb.md

This implementation has been used mostly with Neuftech USB RFID reader, but it also may work with other USB readers as long as they work like a keyboard input.

You could try that, I'm really, if that works.

Darkcast, 2024-10-28:
is there any way I could get this working on the version 2 I know it’s somewhat deprecated, but I noticed how many features are missing in version 3 at the moment. I am willing to try on v3 to test but I am leaning towards v2 at the moment.

s-martin, 2024-10-28:
Reader.py in 2.x ist the USB Reader implementation.

If your reader can be detected as HID / keyboard this could work and you need to register the reader with RegisterDevice.py

Darkcast, 2024-10-28:
Info:
OS: lite os 32 Bookworm 
Phoniebox: V3
Logfile:[INSTALL-1730123804.log](https://github.com/user-attachments/files/17543851/INSTALL-1730123804.log)

Notes:
check that the user was in the input group before running the installer
Run the stable installer, choose generic but the usb didnt show up on the menu
since there is no way to go back to the previous menu I choose the zero option and fake reader just to finish the setup

Let me know if there is anything you would like me to try.