MiczFlor/RPi-Jukebox-RFID #2456: 🐛 | future3/develop installation fails on downloading webapp bundle

pabera, 2024-11-25:
I tried running the installation with future3/develop and I ran into the below error.

2 things need to be fixed
1. Treat future3/develop just like any other developer branch (suggest building of Webapp on device instead of download)
1. The logs don't tell which path was tried to download from. Extend the `validate_url` and `download_from_url` functions to put a log entry with the value.

```bash
#########################################################
Install Web App
  Downloading Web App

****************************************
ERROR OCCURRED!
A non-recoverable error occurred.
Check install log for details:
/home/pi/INSTALL-1732555314.log
****************************************
No prebuild Web App bundle found!
****************************************
```

MaStFAU, 2024-11-28:
I am getting the same on future3/main.

pabera, 2024-11-28:
I will try to get to the bottom of it soon!

pgrete, 2025-01-19:
Is there any news on this? Can I recover manually?