MiczFlor/RPi-Jukebox-RFID #2457: 🐛 | Does not recognize USB RFID-Reader in Web UI

jonasschmid404, 2024-11-25:
### Version

2.8.0 - 5044358f

### Branch

master

### OS

Raspbian GNU/Linux 11 (bullseye)

### Pi model

3 B v1.2

### Hardware

[USB RFID-Reader (Neuftech for EM4100)](https://www.amazon.de/gp/product/B018OYOR3E?&linkCode=sl1&tag=diebarnfeld-21&linkId=f924b3d7972dadb096709b95f757ca72&language=de_DE&ref_=as_li_ss_tl)

### What happened?

Hi everyone, so I'm pretty new to this thing - but very enthusiastic about this project. So I have a problem: when swiping an RFID-card, it is read in the command line interface ("-bash: 00055774785: command not found"), but not in the web UI (javascript is activated. I also can't enter anything manually in the corresponding field "Last Used Chip ID". The issue is somewhat related to issue #768, but not quite.  Thank you for your help, it is greatly appreciated!

s-martin, 2024-11-25:
What’s the output of `tail -n 200 /var/log/syslog`?

jonasschmid404, 2024-11-25:
sorry, I'm too stupid. How can I copy the log from the command line in rasperry?

s-martin, 2024-11-25:
Depends what ssh client you use, but usually you can select the output with your mouse and select all with `CTRL+A` and then copy.

jonasschmid404, 2024-11-25:
I don't have an ssh client set up, I'm watching the raspberry output locally... Ok, so I'll do this first.

jonasschmid404, 2024-11-25:
Here it is... Nov 25 22:42:08 audiobox PhonieboxGPIOControl[27330]:     device = self.generate_device(config[section], section)
Nov 25 22:42:08 audiobox PhonieboxGPIOControl[27330]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/gpio_control.py", line 62, in generate_device
Nov 25 22:42:08 audiobox PhonieboxGPIOControl[27330]:     return SimpleButton(config.getint('Pin'),
Nov 25 22:42:08 audiobox PhonieboxGPIOControl[27330]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/GPIODevices/simple_button.py", line 93, in __init__
Nov 25 22:42:08 audiobox PhonieboxGPIOControl[27330]:     GPIO.add_event_detect(self.pin, edge=self.edge, callback=self.callbackFunctionHandler,
Nov 25 22:42:08 audiobox PhonieboxGPIOControl[27330]: RuntimeError: Conflicting edge detection already enabled for this GPIO channel
Nov 25 22:42:08 audiobox systemd[1]: phoniebox-gpio-control.service: Main process exited, code=exited, status=1/FAILURE
Nov 25 22:42:08 audiobox systemd[1]: phoniebox-gpio-control.service: Failed with result 'exit-code'.
Nov 25 22:42:12 audiobox systemd[1]: phoniebox-gpio-control.service: Scheduled restart job, restart counter is at 617.
Nov 25 22:42:12 audiobox systemd[1]: Stopped Phoniebox GPIO Control Service.
Nov 25 22:42:12 audiobox systemd[1]: Started Phoniebox GPIO Control Service.
Nov 25 22:42:12 audiobox PhonieboxGPIOControl[27371]: INFO:__main__:GPIO Started
Nov 25 22:42:12 audiobox PhonieboxGPIOControl[27371]: INFO:__main__:['VolumeControl', 'RotaryVolumeControl', 'PrevNextControl', 'PlayPause', 'Shutdown', 'PauseShutdown', 'Volume0', 'VolumeUp', 'VolumeDown', 'NextSong', 'PrevSong', 'FastForward', 'Rewind', 'Halt', 'RFIDDevice', 'TriggerPlayCard1', 'TriggerPlayFolderSomeRelativeFolderName', 'StatusLED']
Nov 25 22:42:12 audiobox PhonieboxGPIOControl[27371]: INFO:__main__:Device VolumeControl not enabled
Nov 25 22:42:12 audiobox PhonieboxGPIOControl[27371]: INFO:__main__:adding GPIO-Device, RotaryVolumeControl
Nov 25 22:42:12 audiobox PhonieboxGPIOControl[27371]: INFO:__main__:adding GPIO-Device, PrevNextControl
Nov 25 22:42:12 audiobox PhonieboxGPIOControl[27371]: /home/pi/RPi-Jukebox-RFID/components/gpio_control/GPIODevices/simple_button.py:90: RuntimeWarning: A physical pull up resistor is fitted on this channel!
Nov 25 22:42:12 audiobox PhonieboxGPIOControl[27371]:   GPIO.setup(self.pin, GPIO.IN, pull_up_down=self.pull_up_down)
Nov 25 22:42:12 audiobox PhonieboxGPIOControl[27371]: INFO:GPIODevices.two_button_control:adding new action
Nov 25 22:42:12 audiobox PhonieboxGPIOControl[27371]: INFO:GPIODevices.simple_button:PrevNextControlBtn1: set when_pressed
Nov 25 22:42:12 audiobox PhonieboxGPIOControl[27371]: INFO:GPIODevices.simple_button:add new action
Nov 25 22:42:12 audiobox PhonieboxGPIOControl[27371]: INFO:GPIODevices.simple_button:PrevNextControlBtn2: set when_pressed
Nov 25 22:42:12 audiobox PhonieboxGPIOControl[27371]: INFO:GPIODevices.simple_button:add new action
Nov 25 22:42:12 audiobox PhonieboxGPIOControl[27371]: INFO:__main__:Device PlayPause not enabled
Nov 25 22:42:12 audiobox PhonieboxGPIOControl[27371]: INFO:__main__:adding GPIO-Device, Shutdown
Nov 25 22:42:12 audiobox PhonieboxGPIOControl[27371]: RotaryVolumeControl
Nov 25 22:42:12 audiobox PhonieboxGPIOControl[27371]: PrevNextControl
Nov 25 22:42:12 audiobox PhonieboxGPIOControl[27371]: Shutdown
Nov 25 22:42:12 audiobox PhonieboxGPIOControl[27371]: Traceback (most recent call last):
Nov 25 22:42:12 audiobox PhonieboxGPIOControl[27371]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/gpio_control.py", line 140, in <module>
Nov 25 22:42:12 audiobox PhonieboxGPIOControl[27371]:     devices = gpio_control_class.get_all_devices(config)
Nov 25 22:42:12 audiobox PhonieboxGPIOControl[27371]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/gpio_control.py", line 108, in get_all_devices
Nov 25 22:42:12 audiobox PhonieboxGPIOControl[27371]:     device = self.generate_device(config[section], section)
Nov 25 22:42:12 audiobox PhonieboxGPIOControl[27371]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/gpio_control.py", line 62, in generate_device
Nov 25 22:42:12 audiobox PhonieboxGPIOControl[27371]:     return SimpleButton(config.getint('Pin'),
Nov 25 22:42:12 audiobox PhonieboxGPIOControl[27371]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/GPIODevices/simple_button.py", line 93, in __init__
Nov 25 22:42:12 audiobox PhonieboxGPIOControl[27371]:     GPIO.add_event_detect(self.pin, edge=self.edge, callback=self.callbackFunctionHandler,
Nov 25 22:42:12 audiobox PhonieboxGPIOControl[27371]: RuntimeError: Conflicting edge detection already enabled for this GPIO channel
Nov 25 22:42:12 audiobox systemd[1]: phoniebox-gpio-control.service: Main process exited, code=exited, status=1/FAILURE
Nov 25 22:42:12 audiobox systemd[1]: phoniebox-gpio-control.service: Failed with result 'exit-code'.
Nov 25 22:42:16 audiobox systemd[1]: phoniebox-gpio-control.service: Scheduled restart job, restart counter is at 618.
Nov 25 22:42:16 audiobox systemd[1]: Stopped Phoniebox GPIO Control Service.
Nov 25 22:42:16 audiobox systemd[1]: Started Phoniebox GPIO Control Service.
Nov 25 22:42:16 audiobox PhonieboxGPIOControl[27411]: INFO:__main__:GPIO Started
Nov 25 22:42:16 audiobox PhonieboxGPIOControl[27411]: INFO:__main__:['VolumeControl', 'RotaryVolumeControl', 'PrevNextControl', 'PlayPause', 'Shutdown', 'PauseShutdown', 'Volume0', 'VolumeUp', 'VolumeDown', 'NextSong', 'PrevSong', 'FastForward', 'Re [... cut ...]
Nov 25 22:42:32 audiobox PhonieboxGPIOControl[27575]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/GPIODevices/simple_button.py", line 93, in __init__
Nov 25 22:42:32 audiobox PhonieboxGPIOControl[27575]:     GPIO.add_event_detect(self.pin, edge=self.edge, callback=self.callbackFunctionHandler,
Nov 25 22:42:32 audiobox PhonieboxGPIOControl[27575]: RuntimeError: Conflicting edge detection already enabled for this GPIO channel
Nov 25 22:42:32 audiobox systemd[1]: phoniebox-gpio-control.service: Main process exited, code=exited, status=1/FAILURE
Nov 25 22:42:32 audiobox systemd[1]: phoniebox-gpio-control.service: Failed with result 'exit-code'.

s-martin, 2024-11-25:
Did you try to swipe a card right before you got the log file ?

s-martin, 2024-11-25:
Probably unrelated, something’s wrong with your GPIO settings.
Can you post the content of `~/RPi-Jukebox-RFID/settings/gpio_settings.ini`?

s-martin, 2024-11-26:
Please use `cat ~/RPi-Jukebox-RFID/settings/gpio_settings.ini`

jonasschmid404, 2024-11-26:
[DEFAULT]
enabled: True
antibouncehack: True

[VolumeControl]
enabled: False
Type: TwoButtonControl
Pin1: 5
Pin2: 6
pull_up_down: pull_up
hold_time: 0.3
hold_mode: Repeat
functionCall1: functionCallVolU
functionCall2: functionCallVolD
functionCallTwoButtons: functionCallVol0
;functionCall1Args: 1
;functionCall2Args: 1
;functionCallTwoButtonsArgs: x

[RotaryVolumeControl]
enabled: True
Type: RotaryEncoder
Pin1: 5
Pin2: 6
timeBase: 0.1
functionCall1: functionCallVolU
functionCall2: functionCallVolD

[PrevNextControl]
enabled: False
Type: TwoButtonControl
Pin1: 15
Pin2: 16
functionCall1: functionCallPlayerPrev
functionCall2: functionCallPlayerNext
functionCallTwoButtons: None
pull_up_down: pull_up
hold_time: 0.2
hold_mode: None

[PlayPause]
enabled: False
Type: Button
Pin: 27
pull_up_down: pull_up
functionCall: functionCallPlayerPause

[Shutdown]
enabled: True
Type:  Button
Pin: 3
pull_up_down: pull_up
hold_mode: Postpone
hold_time: 3.0
functionCall: functionCallShutdown

[PauseShutdown]
enabled: False
Type:  Button
Pin: 3
pull_up_down: pull_up
hold_time: 2.0
hold_mode: SecondFunc
functionCall: functionCallPlayerPause
functionCall2: functionCallShutdown

[Volume0]
enabled: False
Type:  Button
Pin: 17
pull_up_down: pull_up
functionCall: functionCallVol0

[VolumeUp]
enabled: False
Type:  Button
Pin: 16
pull_up_down: pull_up
hold_time: 0.3
hold_mode: Repeat
functionCall: functionCallVolU
;functionCallArgs: 1

[VolumeDown]
enabled: False
Type:  Button
Pin: 19
pull_up_down: pull_up
hold_time: 0.3
hold_mode: Repeat
functionCall: functionCallVolD
;functionCallArgs: 1

[NextSong]
enabled: True
Type:  Button
Pin: 16
pull_up_down: pull_up
functionCall: functionCallPlayerNext

[PrevSong]
enabled: True
Type:  Button
Pin: 15
pull_up_down: pull_up
functionCall: functionCallPlayerPrev

[FastForward]
enabled: False
Type:  Button
Pin: 7
pull_up_down: pull_up
functionCall: functionCallPlayerSeekFwd
;functionCallArgs: 10

[Rewind]
enabled: False
Type:  Button
Pin:
pull_up_down: pull_up
functionCall: functionCallPlayerSeekBack
;functionCallArgs: 10

[Halt]
enabled: False
Type:  Button
Pin: 25
pull_up_down: pull_up
functionCall: functionCallPlayerPauseForce

[RFIDDevice]
enabled: False
Type: Button
Pin: 21
pull_up_down: pull_up
functionCall: functionCallPlayerStop

[TriggerPlayCard1]
enabled: False
Type: Button
Pin: 4
pull_up_down: pull_up
functionCall: functionCallTriggerPlayCardId
functionCallArgs: 1

[TriggerPlayFolderSomeRelativeFolderName]
enabled: False
Type: Button
Pin: 4
pull_up_down: pull_up
functionCall: functionCallTriggerPlayFolder
functionCallArgs: someRelativeFolderName

[StatusLED]
enabled: True
Type: StatusLED
Pin: 10

s-martin, 2024-11-26:
You have to make sure that every pin in gpio_settings.ini is defined only once. 

You can remove sections you don't use or at least comment them out with a leading `;`

s-martin, 2024-11-27:
Not sure, if your soundcard is recognized.

please check here for audio troubleshooting: https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#audio-is-not-working