MiczFlor/RPi-Jukebox-RFID #2461: 🐛 | auses during playback after RFID card scan

Manuelibt, 2024-12-02:
### Version

2.8.0

### Branch

master

### OS

Raspbian Bullseye

### Pi model

Raspi 3 b

### What happened?

Hello,

I am experiencing an issue with my Phoniebox installation where the playback is regularly paused, even though the configuration is set up to continue playback as long as the RFID card remains on the reader. It seems that the playback is paused intermittently after the card is detected.

Reproduction:
The Phoniebox is correctly configured to continue playback after an RFID card is detected.
A card is placed on the reader.
After a short period (usually less than a second), the playback is paused, even though the card is still on the reader.
Observed Behavior:
Playback is repeatedly paused after the card is detected.
The process shows in the logs that the RFID reader's signal is not being detected continuously, which leads to the repeated pauses.
Example Log (Excerpt):
plaintext
Code kopieren
2024-12-02 22:37:28,319 - root - INFO - No RFID Signal detected.
2024-12-02 22:37:28,320 - root - INFO - Trigger Pause Force
2024-12-02 22:37:28,319 - root - INFO - Trigger Play Cardid=3495374621
Possible Cause:
The issue might be caused by a delay in the reading process or by the signal handling. The code includes a time.sleep(0.2) delay, which may be interfering with the continuous detection of the RFID signal.

Expected Behavior:
Playback should continue uninterrupted as long as the card remains on the reader.
There should be no unnecessary pauses or delays.
Steps to Reproduce:
Place an RFID card on the reader.
Observe the playback pausing, even though the card is still on the reader.
I have already tried optimizing the configuration and the code, but the issue persists. Can you please assist me in resolving this problem?

Thank you!

s-martin, 2024-12-08:
Which reader do you use?

jonaskor, 2024-12-17:
Same happens for me using the Neuftech USB Card Reader. I therefore disabled the option and just use the "Swiping starts the player"-option. It would have been useful to know that it wouldn't work as expected, I probably would have designed my box differently in that case.