MiczFlor/RPi-Jukebox-RFID #2462: 🐛 |Second Swipe 🚧 This feature is not yet enabled.

jawhett, 2024-12-03:
### Version

3.5.3

### Branch

future3/main 

### OS

VERSION="12 (bookworm)

### Pi model

3B+

### Hardware

PN532 connected through I2C, usb/jack speakers

### What happened?

I dont see anywhere/documentation to fix so might be a bug. But Under the Second Swipe action, it just says " |Second Swipe 🚧 This feature is not yet enabled." How do I enable it?  I see there are some comments on the feature status that something may be wrong but it has the feature checked: "Second Swipe Options -> must be part of player control (partially broken at the moment)"

I dug into the RFID.yaml file and only see a feature for stopping the playback on removal.

the jukebox.yaml file has the following snippet:

  "second_swipe_action:
    # Note: Does not follow the RPC alias convention (yet)
    # Must be one of: 'toggle', 'play', 'skip', 'rewind', 'replay', 'none'
    alias: toggle"
However the current status of a second swipe is it restarts  the file/playlist.

PittBie, 2024-12-07:
I wanted to change the behavior as well. Just recently did a clean install with the develop version. Currently a second swipe of the same RFID tag `rewinds` and plays the playlist from the top.
Would like to change that to Play/Pause.

s-martin, 2024-12-08:
The second swipe feature is currently reworked in https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/2452