MiczFlor/RPi-Jukebox-RFID #2463: 🚀 | Browsing folders in Web App should only work for shared folders or even only audiofolders

MaStFAU, 2024-12-04:
### Feature

When browsing folders in the Web App, going "back" should not go beyond the shared folder

### User perspective

When browsing folders in the Web App, going "back" should not go beyond the shared folder.

pabera, 2024-12-17:
Hey @MaStFAU .

Thanks for reporting this. I remember this problem was already attempted to be fixed in the past. In which situation does it occur? Can you explain the steps you have done to get there?

astnerdev, 2025-11-02:
I can reproduce this: when the back button is pressed, you get back all the way to the home directory.

[image]

[image]