MiczFlor/RPi-Jukebox-RFID #2475: 🐛 | Persistent Max Volume Issue with HiFiBerry on Pi Zero

maganzi, 2024-12-30:
### Version

2.8.0 - 5044358f

### Branch

master

### OS

Raspbian GNU/Linux 11 (bullseye) lite 32bit

### Pi model

Raspberry Zero 2 W

### Hardware

**HiFiBerry MiniAmp**
(VC CS-J420X 10 cm)
(USB RFID Reader) works fine

No additional Switches on GPIO at this time

### What happened?

I encountered a problem with volume control. 
Audio playback works, but the volume cannot be adjusted and it is persistently maxed out and extremely loud.
The volume displayed in **Settings->Select volume manager** **mpd** changes when I attempt to adjust it either via the web interface or using **amixer** commands, yet these changes do not affect the actual audio output. Pressing 'set' in amixer does nothing — the volume remains constant and unresponsive. 

### Configuration

sudo nano /etc/mpd.conf

...

# Audio Output ################################################################
#
# MPD supports various audio output types, as well as playing through multiple
# audio outputs at the same time, through multiple audio_output settings
# blocks. Setting this block is optional, though the server will only attempt
# autodetection for one sound card.
#
# An example of an ALSA output:
#
audio_output {
type            "alsa"
        name            "My ALSA Device"
#       device          "hw:0,0"        # optional
#       mixer_type      "hardware"      # optional
#       mixer_device    "default"       # optional
#        mixer_control   "PCM"           # optional
#       mixer_index     "0"             # optional
}
#

...

boot/config.txt

# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
**# dtparam=audio=on**

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Enable DRM VC4 V3D driver
**dtoverlay=vc4-kms-v3d,noaudio**
max_framebuffers=2

# Disable compensation for displays with overscan
disable_overscan=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

[pi4]
# Run as fast as firmware / board allows
arm_boost=1

[all]

**dtoverlay=hifiberry-dac
force_eeprom_read=0**

### More info

Thank you for this fantastic repository! I hope someone can assist us in resolving this issue. Thank you in advance!

AlvinSchiller, 2024-12-30:
Hi, can you try setting the `mixer_type` from `hardware` to `software`? Don't forget to activate the entry by renovating the leading "#".
See https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/654#issuecomment-562990679

maganzi, 2024-12-30:
I apologize for my earlier message; it seems I completely overlooked the impact of your suggestion. After setting the mixer_type to 'software', everything works perfectly now. Thank you very much for your help!

closed 2024-12-30 by maganzi