MiczFlor/RPi-Jukebox-RFID #2483: 🐛 | missing gpio.yaml in folder /settings after installation

Sankara49, 2025-01-17:
### Version

3.5.3

### Branch

future3/main

### OS

RPi OS Bullseye 32bit lite

### Pi model

Raspberry Pi 3 Model B V1.2

### Hardware

- Neuftech USB RFID reader
- GPIO Arcade Button (want to)
- USB speaker

### What happened?

1. stop jukebox service - success
2. enable gpioz in jukebox.yaml -success
3. run_jukebox.sh - **Error look at "logs":  file gpio.yaml not Found -> disable GPIOZ**
4. restart service - not tried

### Logs

17.01.2025 11:00:42 -  341:__init__.py        - jb.gpioz             - MainThread      - ERROR    - Disable GPIOZ due to error loading GPIOZ config file. FileNotFoundError: [Errno 2] No such file or directory: '../../shared/settings/gpio.yaml'

### More info

Hello,

- I tried to install the GPIOZ files yesterday evening manually but was not able to find the corresponding install.sh file in(respectively or) any gpio folders.
- today I did a fresh installation to find out, if this will fix the missing gpio.yaml in /shared/settings
- found out, that it still not exist and that I am still not able to find any instlall.sh for installing gpio manually

I am looking forward someone can help me or tell me a workaround.
greetings Sankara

Gen-Chan, 2025-01-18:
Hey there. If I get you right, I think the reason for the file "missing" is, that the feature optionally. In the [documentation](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/future3/main/documentation/builders/gpio.md) is a part about the GPIO. As you already done, you need to enable it in the `jukebox.yaml` and then create a `gpio.yaml` by your own. An example file can be found `resources/default-settings/gpio.yaml` or you create your own from scratch.

s-martin, 2025-01-19:
Yes, that’s correct you need to add that file, if you want to use that feature.