MiczFlor/RPi-Jukebox-RFID #2489: After reboot persistent Alsa error

deepblue897, 2025-02-01:
### Version

2.9.0-alpha - 4816b1f8 - develop

### Branch

develop

### OS

Raspbian GNU/Linux 12 (bookworm)

### Pi model

Pi 4

### What happened?

System was running fine (music plays, GPIO buttons worked) and then I enabled an additional GPIO pin.
After reboot no music could be played any more. 

mpd throws errors about ALSA device.
Feb 01 18:07:02 phoniebox mpd[737]: exception: Failed to open "My ALSA Device" (alsa); Failed to open ALSA device "default": Unknown error 524
Feb 01 18:07:02 phoniebox mpd[737]: exception: Failed to open "My ALSA Device" (alsa); Failed to open ALSA device "default": Unknown error 524

After restoring the settings in the gpio config (replace entire file by backup copy) & reboot the error persists

### Logs

[log.txt](https://github.com/user-attachments/files/18629706/log.txt)

s-martin, 2025-02-04:
Probably unrelated, but the USB encoder does not work

```
Feb 01 18:29:01 phoniebox systemd[1]: Started phoniebox-buttons-usb-encoder.service - Phoniebox Buttons USB Encoder Service.
Feb 01 18:29:02 phoniebox buttons_usb_encoder.py[4969]: Please run map_buttons_usb_encoder.py first
```

Follow the instructions here https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/develop/components/controls/buttons_usb_encoder#installation

s-martin, 2025-02-04:
Please check this discussion for the ALSA issue https://github.com/MiczFlor/RPi-Jukebox-RFID/discussions/2267