MiczFlor/RPi-Jukebox-RFID #2496: 🐛 |  Permission denied: '/dev/i2c-1'

mastin-zgz, 2025-02-10:
### Version

2.8.0 - 5044358f - master

### Branch

master

### OS

"Raspbian GNU/Linux 11 (bullseye)" 32bits

### Pi model

3B

### Hardware

RFID PN532

### What happened?

Good morning,  

I have installed the entire system from scratch in a clean way, but the RID reader, although it was detected during installation, gives me this error when running it: "the journaling -xe".  
``
Any ideas?

### Logs

Feb 10 11:46:08 musica daemon_rfid_reader.py[1589]: PermissionError: [Errno 13] Permission denied: '/dev/i2c-1'
Feb 10 11:46:09 musica daemon_rfid_reader.py[1599]: Traceback (most recent call last):
Feb 10 11:46:09 musica daemon_rfid_reader.py[1599]:   File "/home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py", line 20, in <module>
Feb 10 11:46:09 musica daemon_rfid_reader.py[1599]:     reader = Reader()
Feb 10 11:46:09 musica daemon_rfid_reader.py[1599]:   File "/home/pi/RPi-Jukebox-RFID/scripts/Reader.py", line 221, in __init__
Feb 10 11:46:09 musica daemon_rfid_reader.py[1599]:     self.reader = Pn532Reader()
Feb 10 11:46:09 musica daemon_rfid_reader.py[1599]:   File "/home/pi/RPi-Jukebox-RFID/scripts/Reader.py", line 191, in __init__
Feb 10 11:46:09 musica daemon_rfid_reader.py[1599]:     pn532 = Pn532_i2c() # noqa F841
Feb 10 11:46:09 musica daemon_rfid_reader.py[1599]:   File "/usr/local/lib/python3.9/dist-packages/py532lib/i2c.py", line 56, in __init__
Feb 10 11:46:09 musica daemon_rfid_reader.py[1599]:     self.PN532 = I2CMaster(self.i2c_channel)
Feb 10 11:46:09 musica daemon_rfid_reader.py[1599]:   File "/usr/local/lib/python3.9/dist-packages/quick2wire/i2c.py", line 48, in __init__
Feb 10 11:46:09 musica daemon_rfid_reader.py[1599]:     self.fd = posix.open("/dev/i2c-%i"%n, posix.O_RDWR|extra_open_flags)
Feb 10 11:46:09 musica daemon_rfid_reader.py[1599]: PermissionError: [Errno 13] Permission denied: '/dev/i2c-1'
Feb 10 11:46:10 musica daemon_rfid_reader.py[1608]: Traceback (most recent call last):
Feb 10 11:46:10 musica daemon_rfid_reader.py[1608]:   File "/home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py", line 20, in <module>
Feb 10 11:46:10 musica daemon_rfid_reader.py[1608]:     reader = Reader()
Feb 10 11:46:10 musica daemon_rfid_reader.py[1608]:   File "/home/pi/RPi-Jukebox-RFID/scripts/Reader.py", line 221, in __init__
Feb 10 11:46:10 musica daemon_rfid_reader.py[1608]:     self.reader = Pn532Reader()
Feb 10 11:46:10 musica daemon_rfid_reader.py[1608]:   File "/home/pi/RPi-Jukebox-RFID/scripts/Reader.py", line 191, in __init__
Feb 10 11:46:10 musica daemon_rfid_reader.py[1608]:     pn532 = Pn532_i2c() # noqa F841
Feb 10 11:46:10 musica daemon_rfid_reader.py[1608]:   File "/usr/local/lib/python3.9/dist-packages/py532lib/i2c.py", line 56, in __init__
Feb 10 11:46:10 musica daemon_rfid_reader.py[1608]:     self.PN532 = I2CMaster(self.i2c_channel)
Feb 10 11:46:10 musica daemon_rfid_reader.py[1608]:   File "/usr/local/lib/python3.9/dist-packages/quick2wire/i2c.py", line 48, in __init__
Feb 10 11:46:10 musica daemon_rfid_reader.py[1608]:     self.fd = posix.open("/dev/i2c-%i"%n, posix.O_RDWR|extra_open_flags)
Feb 10 11:46:10 musica daemon_rfid_reader.py[1608]: PermissionError: [Errno 13] Permission denied: '/dev/i2c-1'
Feb 10 11:46:11 musica daemon_rfid_reader.py[1611]: Traceback (most recent call last):
Feb 10 11:46:11 musica daemon_rfid_reader.py[1611]:   File "/home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py", line 20, in <module>
Feb 10 11:46:11 musica daemon_rfid_reader.py[1611]:     reader = Reader()
Feb 10 11:46:11 musica daemon_rfid_reader.py[1611]:   File "/home/pi/RPi-Jukebox-RFID/scripts/Reader.py", line 221, in __init__
Feb 10 11:46:11 musica daemon_rfid_reader.py[1611]:     self.reader = Pn532Reader()
Feb 10 11:46:11 musica daemon_rfid_reader.py[1611]:   File "/home/pi/RPi-Jukebox-RFID/scripts/Reader.py", line 191, in __init__
Feb 10 11:46:11 musica daemon_rfid_reader.py[1611]:     pn532 = Pn532_i2c() # noqa F841
Feb 10 11:46:11 musica daemon_rfid_reader.py[1611]:   File "/usr/local/lib/python3.9/dist-packages/py532lib/i2c.py", line 56, in __init__
Feb 10 11:46:11 musica daemon_rfid_reader.py[1611]:     self.PN532 = I2CMaster(self.i2c_channel)
Feb 10 11:46:11 musica daemon_rfid_reader.py[1611]:   File "/usr/local/lib/python3.9/dist-packages/quick2wire/i2c.py", line 48, in __init__
Feb 10 11:46:11 musica daemon_rfid_reader.py[1611]:     self.fd = posix.open("/dev/i2c-%i"%n, posix.O_RDWR|extra_open_flags)
Feb 10 11:46:11 musica daemon_rfid_reader.py[1611]: PermissionError: [Errno 13] Permission denied: '/dev/i2c-1'

AlvinSchiller, 2025-02-10:
Hi, did you use the installer for the RFID reader or did you manually set it up by documentation?

mastin-zgz, 2025-02-10:
Hi,
I do a fresh install with te documentation.

Anyway, I'm looking to have version 2 on my Raspberry Pi 3 with GPIO, but I can't find the right combination of system and kernel that makes everything work. Any ideas? Which version of Raspbian should I download with which kernel?

Thanks!