MiczFlor/RPi-Jukebox-RFID #2503: 🐛 | run_register_rfid_reader.py broken in stable brunch

rudynz, 2025-02-25:
### Version

3.5.3

### Branch

future3/main

### OS

Raspbian GNU/Linux 11 (bullseye)"

### Pi model

Zero 2

### Hardware

neuftech USB Reader

### What happened?

The USB Configuration scrips fails:

pi@selma-musika:~/RPi-Jukebox-RFID/installation/components $ ./setup_rfid_reader.sh
Choose Reader Module from list:

  0: No RFID Reader                           (Module: No RFID Reader)
  1: Generic USB Reader                       (Module: generic_usb/generic_usb.py)
  2: RDM6300 via serial UART                  (Module: rdm6300_serial/rdm6300_serial.py)
  3: MFRC522 via SPI                          (Module: rc522_spi/rc522_spi.py)
  4: PN532 reader via I2C using py532 library (Module: pn532_i2c_py532/pn532_i2c_py532.py)
  5: Fake Reader (simulator for development)  (Module: fake_reader_gui/fake_reader_gui.py)
  6: Generic NFCPY NFC Reader Module          (Module: generic_nfcpy/generic_nfcpy.py)

Reader module number? [0..6] 1
Reader selected: 'generic_usb'

Entering reader customization

Choose RFID device from USB device list:
If you are planning to connect multiple USB Readers make sure ALL of them are connected before running this script!If your RFID reader appears multiple times (e.g. KKMoon), select the one which has PASS in the isKey column.
For the curious: isKey indicates a device capability check for required keys
Traceback (most recent call last):
  File "/home/pi/RPi-Jukebox-RFID/src/jukebox/run_register_rfid_reader.py", line 72, in <module>
    main()
  File "/home/pi/RPi-Jukebox-RFID/src/jukebox/run_register_rfid_reader.py", line 67, in main
    rfid_configure.query_user_for_reader(dependency_install=dinstall_lookup[args.deps]),
  File "/home/pi/RPi-Jukebox-RFID/src/jukebox/components/rfid/configure/__init__.py", line 214, in query_user_for_reader
    reader_params = reader_module.query_customization()
  File "/home/pi/RPi-Jukebox-RFID/src/jukebox/components/rfid/hardware/generic_usb/generic_usb.py", line 81, in query_customization
    devices_is_key = [_is_keyboard(x) for x in devices]
  File "/home/pi/RPi-Jukebox-RFID/src/jukebox/components/rfid/hardware/generic_usb/generic_usb.py", line 81, in <listcomp>
    devices_is_key = [_is_keyboard(x) for x in devices]
  File "/home/pi/RPi-Jukebox-RFID/src/jukebox/components/rfid/hardware/generic_usb/generic_usb.py", line 69, in _is_keyboard
    logger.debug(f"is_keyboard test for '{device.name}' at '{device.fn}' is '{is_keyboard_res}'")
AttributeError: 'InputDevice' object has no attribute 'fn'

### More info

I copied the file "generic_usb.py" version from the development brunch and then it worked.
    /src/jukebox/components/rfid/hardware/generic_usb/

Th eproblem seem to be in line 69
  logger.debug(f"is_keyboard test for '{device.name}' at '{device.fn}' is '{is_keyboard_res}'")

 logger.debug(f"is_keyboard test for '{device.name}' at '{device.path}' is '{is_keyboard_res}'")

AlvinSchiller, 2025-02-26:
As you already described was this fixes with #2497. A release is coming soon to get this on the stable branch.

s-martin, 2025-05-28:
Duplicate of #2497