MiczFlor/RPi-Jukebox-RFID #2515: 🐛 | RFID service won't start

Momro, 2025-03-30:
### Version

2.8.0 - 5044358f - master

### Branch

master

### OS

RPi 64bit lite

### Pi model

Zero 2W

### Hardware

RC522 reader

### What happened?

I am trying to run RegisterDevice.py, but it fails

### Logs

# journalctl -n 200 | catcat
Mar 30 22:31:24 phoniebox systemd[1]: phoniebox-buttons-usb-encoder.service: Scheduled restart job, restart counter is at 15.
Mar 30 22:31:24 phoniebox systemd[1]: Stopped phoniebox-buttons-usb-encoder.service - Phoniebox Buttons USB Encoder Service.
Mar 30 22:31:24 phoniebox (coder.py)[1622]: phoniebox-buttons-usb-encoder.service: Failed to locate executable /home/pi/RPi-Jukebox-RFID/components/controls/buttons_usb_encoder/buttons_usb_encoder.py: Permission denied
Mar 30 22:31:24 phoniebox (coder.py)[1622]: phoniebox-buttons-usb-encoder.service: Failed at step EXEC spawning /home/pi/RPi-Jukebox-RFID/components/controls/buttons_usb_encoder/buttons_usb_encoder.py: Permission denied
Mar 30 22:31:24 phoniebox systemd[1]: Started phoniebox-buttons-usb-encoder.service - Phoniebox Buttons USB Encoder Service.
Mar 30 22:31:24 phoniebox systemd[1]: phoniebox-buttons-usb-encoder.service: Main process exited, code=exited, status=203/EXEC
Mar 30 22:31:24 phoniebox systemd[1]: phoniebox-buttons-usb-encoder.service: Failed with result 'exit-code'.
Mar 30 22:31:28 phoniebox sudo[1624]: www-data : PWD=/home/pi/RPi-Jukebox-RFID/htdocs ; USER=root ; COMMAND=/usr/bin/atq -q s
Mar 30 22:31:28 phoniebox sudo[1624]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=33)
Mar 30 22:31:28 phoniebox sudo[1624]: pam_unix(sudo:session): session closed for user root
Mar 30 22:31:28 phoniebox sudo[1628]: www-data : PWD=/home/pi/RPi-Jukebox-RFID/htdocs ; USER=root ; COMMAND=/usr/bin/atq -q t
Mar 30 22:31:28 phoniebox sudo[1628]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=33)
Mar 30 22:31:28 phoniebox sudo[1628]: pam_unix(sudo:session): session closed for user root
Mar 30 22:31:28 phoniebox sudo[1632]: www-data : PWD=/home/pi/RPi-Jukebox-RFID/htdocs ; USER=root ; COMMAND=/usr/bin/atq -q q
Mar 30 22:31:28 phoniebox sudo[1632]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=33)
Mar 30 22:31:28 phoniebox sudo[1632]: pam_unix(sudo:session): session closed for user root
Mar 30 22:31:28 phoniebox kernel: warning: `iwconfig' uses wireless extensions which will stop working for Wi-Fi 7 hardware; use nl80211
Mar 30 22:31:28 phoniebox sudo[1640]: www-data : PWD=/home/pi/RPi-Jukebox-RFID/htdocs ; USER=root ; COMMAND=/usr/bin/bash -c 'source /home/pi/RPi-Jukebox-RFID/scripts/helperscripts/inc.networkHelper.sh && get_wireless_networks'
Mar 30 22:31:28 phoniebox sudo[1640]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=33)
Mar 30 22:31:29 phoniebox sudo[1659]:     root : PWD=/home/pi/RPi-Jukebox-RFID/htdocs ; USER=root ; COMMAND=/usr/bin/nmcli --show-secrets -t -f 802-11-wireless.ssid,802-11-wireless-security.psk,connection.autoconnect-priority con show bb99eb15-0eb4-4007-a4e9-c73529965510
Mar 30 22:31:29 phoniebox sudo[1659]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=0)
Mar 30 22:31:29 phoniebox sudo[1659]: pam_unix(sudo:session): session closed for user root
Mar 30 22:31:29 phoniebox sudo[1640]: pam_unix(sudo:session): session closed for user root
Mar 30 22:31:30 phoniebox sudo[1683]: www-data : PWD=/home/pi/RPi-Jukebox-RFID/htdocs/api ; USER=root ; COMMAND=/usr/bin/echo playlistinfo#012close
Mar 30 22:31:30 phoniebox sudo[1683]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=33)
Mar 30 22:31:30 phoniebox sudo[1683]: pam_unix(sudo:session): session closed for user root
Mar 30 22:31:34 phoniebox systemd[1]: phoniebox-buttons-usb-encoder.service: Scheduled restart job, restart counter is at 16.
Mar 30 22:31:34 phoniebox systemd[1]: Stopped phoniebox-buttons-usb-encoder.service - Phoniebox Buttons USB Encoder Service.
Mar 30 22:31:34 phoniebox (coder.py)[1686]: phoniebox-buttons-usb-encoder.service: Failed to locate executable /home/pi/RPi-Jukebox-RFID/components/controls/buttons_usb_encoder/buttons_usb_encoder.py: Permission denied
Mar 30 22:31:34 phoniebox (coder.py)[1686]: phoniebox-buttons-usb-encoder.service: Failed at step EXEC spawning /home/pi/RPi-Jukebox-RFID/components/controls/buttons_usb_encoder/buttons_usb_encoder.py: Permission denied
Mar 30 22:31:34 phoniebox systemd[1]: Started phoniebox-buttons-usb-encoder.service - Phoniebox Buttons USB Encoder Service.
Mar 30 22:31:34 phoniebox systemd[1]: phoniebox-buttons-usb-encoder.service: Main process exited, code=exited, status=203/EXEC
Mar 30 22:31:34 phoniebox systemd[1]: phoniebox-buttons-usb-encoder.service: Failed with result 'exit-code'.
Mar 30 22:31:38 phoniebox sudo[1704]: www-data : PWD=/home/pi/RPi-Jukebox-RFID/htdocs ; USER=root ; COMMAND=/usr/bin/vcgencmd measure_temp
Mar 30 22:31:38 phoniebox sudo[1704]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=33)
Mar 30 22:31:38 phoniebox sudo[1704]: pam_unix(sudo:session): session closed for user root
Mar 30 22:31:38 phoniebox sudo[1707]: www-data : PWD=/home/pi/RPi-Jukebox-RFID/htdocs ; USER=root ; COMMAND=/usr/bin/vcgencmd get_throttled
Mar 30 22:31:38 phoniebox sudo[1707]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=33)
Mar 30 22:31:38 phoniebox sudo[1707]: pam_unix(sudo:session): session closed for user root
Mar 30 22:31:38 phoniebox sudo[1717]: www-data : PWD=/home/pi/RPi-Jukebox-RFID/htdocs ; USER=root ; COMMAND=/home/pi/RPi-Jukebox-RFID/scripts/inc.writeGlobalConfig.sh
Mar 30 22:31:38 phoniebox sudo[1717]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=33)
Mar 30 22:31:38 phoniebox sudo[1802]:     root : PWD=/home/pi/RPi-Jukebox-RFID/htdocs ; USER=root ; COMMAND=/usr/bin/chown -R pi:www-data /home/pi/RPi-Jukebox-RFID/scripts/../settings/global.conf
Mar 30 22:31:38 phoniebox sudo[1802]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=0)
[... 159 lines cut ...]

I don't think there is a relevant config for the RFID reader. SPI is enabled according to raspi-config.

### More info

- The reader is working. I tested it on a different Pi with a simple script and could read several RFID cards
- Fresh install of OS, no other software installed (Vim and ZSH)
- everything up to date (apt update && apt upgrade && apt dist-upgrade && rpi-update
- moving Reader.py to Reader.py.experimental did not improve anything (ln -s Reader.py Reader.py.experimental). I get the following error when I try to choose my reader from the list:

```
# python3 RegisterDevice.py
Choose the reader from list
0 vc4-hdmi HDMI Jack
1 vc4-hdmi
2 MFRC522
3 RDM6300
4 PN532
Device Number: 2
Traceback (most recent call last):
  File "/home/pi/RPi-Jukebox-RFID/scripts/RegisterDevice.py", line 21, in <module>
    dev.close()
    ^^^^^^^^^
AttributeError: 'NonUsbDevice' object has no attribute 'close'
```

Momro, 2025-04-05:
I ordered the Neuftech USB reader and that one does not work either. Same error message:

```
# python3 RegisterDevice.py
Choose the reader from list
0 vc4-hdmi HDMI Jack
1 vc4-hdmi
2 CM109 USB driver
3 MFRC522
4 RDM6300
5 PN532
Device Number: 2
Traceback (most recent call last):
  File "/home/pi/RPi-Jukebox-RFID/scripts/RegisterDevice.py", line 21, in <module>
    dev.close()
    ^^^^^^^^^
AttributeError: 'NonUsbDevice' object has no attribute 'close'
```

Neither with `Reader.py` nor `Reader.py.experimental` would it work.

willemOH, 2025-06-18:
Wow I was losing it over here with reinstalls trying to get my PN532 (different than the two you tried, im reading) to register cards and show the latest id in the web app. Commenting out `dev.close()` in `RegisterDevice.py` and then restarting the service as the readme instructs - `RPi-Jukebox-RFID\components\rfid-reader\PN532\Readme.md` actually worked. now I can see the id updating in the web app when I swipe a tag. Whats misleading is if you search my chip reader online you will get a thread that verifies this works ["out of the box"](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/931) so clearly some thing(s) have been broken since then. Now to address mpd not being able to configure 🙄

Momro, 2025-06-18:
Happy it works for you, mate :-)
If I was still working on the Phoniebox, I'd order a PN532, but I swapped to "Tonuino" and am about to finish the project ... hopefully ... but might return some time to Phoniebox. I like the Wifi idea ...

Coernel82, 2025-06-18:
I have the same problem, however I already made one big mistake. As the release says:

> [!CAUTION]
> Do NOT use the latest RaspberryPi OS Bookworm release (also don't upgrade via apt) with the new kernel 6.6, if you need to use GPIO components . There have been incompatible changes for the RPi.GPIO library, so Shims and Rfid Reader usage can be broken ([#2295](https://github.com/MiczFlor/RPi-Jukebox-RFID/discussions/2295)).
>
> **Using RaspberryPi OS _Bullseye_ is the current recommendation as it does not use the new kernel.**

I am reinstalling with an old Pi OS and report back...

Momro, 2025-06-20:
I wonder why they did not include a small script in the install process that checks for the current OS, and halts if it is the wrong version.

Very little effort, huge effect

Momro, 2025-07-13:
The repository says I have to be a contributor. I skimmed over the documentation and found nothing on how to become one.

I added this code to the installer script and it works on my Pi Zero with Bookworm installed, see example execution below:

´´´
# cat install-jukebox.sh | grep checkPre -A 10
checkPrerequisite() {
        # Currently, RPi OS Bookworm is not supported, see https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/INSTALL#installing-phoniebox
        # Exit installer if Bookworm is installed
        if [[ $(cat /etc/os-release | grep "VERSION_CODENAME=bookworm" ) != '' ]] ; then
                echo "[ERROR] You are using RPi OS Bookworm. This is not supported by Phoniebox, see install instructions"
                echo "Exit"
                exit 2
        fi

    #currently the user 'pi' is mandatory
    #https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/1785

# cat /etc/os-release                                                                                                                                                                      PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
´´´

s-martin, 2025-07-13:
> The repository says I have to be a contributor. I skimmed over the documentation and found nothing on how to become one.

You have to make a copy of the repo and then open a PR to this repo.
See details here: https://docs.github.com/en/get-started/exploring-projects-on-github/contributing-to-a-project

To contribute directly to the repo, you need to be a contributor.

> I added this code to the installer script and it works on my Pi Zero with Bookworm installed, see example execution below:
> 
> ```
> # cat install-jukebox.sh | grep checkPre -A 10
> checkPrerequisite() {
>         # Currently, RPi OS Bookworm is not supported, see https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/INSTALL#installing-phoniebox
>         # Exit installer if Bookworm is installed
>         if [[ $(cat /etc/os-release | grep "VERSION_CODENAME=bookworm" ) != '' ]] ; then
>                 echo "[ERROR] You are using RPi OS Bookworm. This is not supported by Phoniebox, see install instructions"
>                 echo "Exit"
>                 exit 2
>         fi
> 
>     #currently the user 'pi' is mandatory
>     #https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/1785
> 
> # cat /etc/os-release                                                                                                                                                                      PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
> NAME="Debian GNU/Linux"
> VERSION_ID="12"
> VERSION="12 (bookworm)"
> VERSION_CODENAME=bookworm
> ID=debian
> HOME_URL="https://www.debian.org/"
> SUPPORT_URL="https://www.debian.org/support"
> BUG_REPORT_URL="https://bugs.debian.org/"
> 
> # ./install-jukebox.sh
> [ERROR] You are using RPi OS Bookworm. This is not supported by Phoniebox, see install instructions
> Exit
> ```

Thanks for the suggestion! 🙏🏼