MiczFlor/RPi-Jukebox-RFID #2516: 🐛 | Wifi SSID with space not settable

grandmastermarco, 2025-04-01:
### Version

2.8.0

### Branch

master

### OS

Paspberry Pi Os Lite Legacy (bullseye) 32-bit lite

### Pi model

Zero 2W

### What happened?

i set my home wifi in settings / wifi.

my wifi SSID has several spaces (eg.: "Test 123")
during flashing of pi, i was able to set the wifi correctly with the pi imager. 

after adding another wifi in the UI, the original wifi had no spaces anymore. setting the first wifi again was only possible by adjusting wpa_supplicant.conf. any changes in the UI in the wifi tab remove the needed spaces.

maybe wifi SSID is not stored correctly?
any help is appreciated.

Momro, 2025-04-07:
I think the error is here:

```
# grep "foreach" htdocs/inc.setWifi.php -n
16:    foreach ( $tempPOST as $post_key => $post_value ) {
```

It will go through each string, and if there is a space in between, it will assume that it is two entities.

I'm not a dev here, so I cannot suggest a fix.

According to inc.networkHelper.sh, it should store all Wifis in `/etc/wpa_supplicant/wpa_supplicant.conf`, but I don't see that file in my environment.

You could try to add the Wifi manually there instead of using the GUI on the web interface

jonaskor, 2025-12-03:
I have the same problem. Unfortunately editing the wpa_supplicant.conf also didn`t allow me to reconnect to the wifi with a space in the SSID (file content missed about the second half of the content). A fox would be appreciated!